adafruit/Adafruit_BusIO #7: SPI_BITORDER_MSBFIRST not defined for STM32 Arduino

ElPercha, 2019-11-06:
Hello folks 
This is my first post in Github and I have no idea what I'm doing :)
But I found an issue trying to compile an Arduino  project that uses the "Adafruit MPU6050" Library.
The target device is a STMF103C8 (typical bluepill with 64K) and for the board "Hardware" I'm using:
https://github.com/rogerclarkmelbourne/Arduino_STM32

I got the error "SPI_BITORDER_MSBFIRST not defined"

I fixed it modifying the file Adafruit_SPIDevice.h but I don't know how to commit the change neither if the change is correct.

// Some platforms have a BitOrder enum but its named MSBFIRST/LSBFIRST
#if defined(ARDUINO_ARCH_SAMD) || defined(__SAM3X8E__) || defined(NRF52_SERIES) || defined(ARDUINO_ARCH_ARDUINO_CORE_STM32) || defined(ARDUINO_ARCH_MEGAAVR) **|| defined(__STM32__)**
  #define SPI_BITORDER_MSBFIRST MSBFIRST
  #define SPI_BITORDER_LSBFIRST LSBFIRST
#endif

Hope this is useful to someone else...

ladyada, 2019-11-06:
hi please submit a PR  so others can benefit :)

ElPercha, 2019-11-06:
I just modified my post, it had an error
PR = Pull Request ?

ladyada, 2019-11-06:
yeah! we have a guide on how to submit a PR, this is a pretty easy PR so it's an excellent first one :)

https://learn.adafruit.com/an-introduction-to-collaborating-with-version-control

ElPercha, 2019-11-06:
ok, let me try

ElPercha, 2019-11-06:
Seems like I cannot create a Branch
I red some documentation and looks like I have to create a Branch, make my changes and then create the PR. Is there another way to create the PR?
I tried to create the Branch from the Web and also from GitHub Desktop and seems like I have no permissions to create the branch...

ElPercha, 2019-11-06:
Finally I guess that I figured it out.

PR created :)

PR #8 (2019-11-06): Update Adafruit_SPIDevice.h
Fixing "SPI_BITORDER_MSBFIRST not defined for STM32" When compiling for Arduino / STM32

Trying to fix #7
closed 2019-11-06 by ladyada