adafruit/Adafruit_BusIO #18: Hang on arduino nano IOT device with Adafruit_MAX31855

digitaldan123, 2020-04-15:
- Arduino board: 
Nano 33 IOT 

- Arduino IDE version (found in Arduino -> About Arduino menu): 
  1.8.7

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): 

Using the Adafruit_MAX31855 sample code, serialthermocouple.ino, I was getting a hang trying to read the thermocouple.readInternal() value. On a different Arduino device (Mega ADK) it was working fine. With some debugging I found that I had to wrap the following digitalWrite to not get called if _mosi was -1 (which it is when using the Adafruit_MAX31855.cpp helper class) : https://github.com/adafruit/Adafruit_BusIO/blob/master/Adafruit_SPIDevice.cpp#L127

ladyada, 2020-04-19:
ooh ok yah @siddacious please add a check if _miso is -1 before trying to write and ill review the PR! you dont need an arduino nano to test, use any arduino board.

ladyada, 2020-04-21:
should be resolved with latest commit!

closed 2020-04-21 by ladyada