adafruit/Adafruit_BusIO #19: Define the meaning of 'type' for BusIO Registers

jayjlawrence, 2020-04-19:
Not an issue but hopefully helpful input. I didn't know where best to put this info so am providing here. I struggled a bit over what "type" meant when defining a Adafruit_BusIO_Register and then it clicked once I saw the code. To assist future developers here is an attempt at a couple of definitions:

ADDRBIT8_HIGH_TOREAD When reading a register you must actually send the value 0x80 + register to the device. e.g. To read the register 0x0B the register value 0x8B is sent and to write 0x0B is sent.

ADDRBIT8_HIGH_TOWRITE When writing to a register you must actually send the value 0x80 + register to the device. e.g. To write to the register 0x19 the register value 0x99 is sent and to read 0x19 is sent.

Hope this helps. Also can the documentation be generated and included in the git repository?

ladyada, 2020-04-19:
yep! if you'd like to add that text to the register consts, it will automatically get doxygen'd by Actions and put here:
https://adafruit.github.io/Adafruit_BusIO/html/annotated.html

caternuson, 2021-04-01:
@jayjlawrence Did #41 fix this?

caternuson, 2021-06-17:
Closing due to lack of response. Seems fixed.

closed 2021-06-17 by caternuson

TomD53, 2022-12-18:
I'm also currently staring at the register types. Trying to write my own driver for the ADXL375 and struggling to write data to the registers. It appears that the AD8_HIGH_TOREAD_AD7_HIGH_TOINC is used. 

What does this mean? Can someone provide a text description similar to above? Thanks