adafruit/Adafruit_BusIO #27: Issue with version 1.2.0

soothsayer011, 2020-05-22:
This library is used in Marlin 2.0 firmware for 3D printing, and since the changes made to Adafruit_MAX31865 yesterday from 1.1.0 to 1.2.0, the firmware no longer compiles. Here is a thread on the issue https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/257

The firmware will compile with version 1.1.0.

Here is the output of the error which references the library
ompiling .pio\build\STM32F103RC_btt_512K\liba17\USBComposite\Consumer.cpp.o
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp: In member function 'bool Adafruit_I2CDevice::read(uint8_t*, size_t, bool)':
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp:160:79: error: no matching function for call to 'TwoWire::requestFrom(uint8_t, uint8_t, uint8_t)'
size_t recv = _wire->requestFrom((uint8_t)_addr, (uint8_t)len, (uint8_t)stop);
^
In file included from C:\Users\lastp.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/Wire.h:42:0,
from .pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214/Adafruit_I2CDevice.h:1,
from .pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp:1:
C:\Users\lastp.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/utility/WireBase.h:101:11: note: candidate: uint8 WireBase::requestFrom(uint8, int)
uint8 requestFrom(uint8, int);
^~~~~~~~~~~
C:\Users\lastp.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/utility/WireBase.h:101:11: note: candidate expects 2 arguments, 3 provided
C:\Users\lastp.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/utility/WireBase.h:106:11: note: candidate: uint8 WireBase::requestFrom(int, int)
uint8 requestFrom(int, int);
^~~~~~~~~~~
C:\Users\lastp.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/utility/WireBase.h:106:11: note: candidate expects 2 arguments, 3 provided
*** [.pio\build\STM32F103RC_btt_512K\lib2ab\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp.o] Error 1
In file included from .pio\libdeps\STM32F103RC_btt_512K\Adafruit_MAX31865\Adafruit_MAX31865.h:55:0,
from .pio\libdeps\STM32F103RC_btt_512K\Adafruit_MAX31865\Adafruit_MAX31865.cpp:17:
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:37:43: error: 'SPI_BITORDER_MSBFIRST' was not declared in this scope
BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
^~~~~~~~~~~~~~~~~~~~~
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:37:43: note: suggested alternative: 'SPI_CR1_LSBFIRST'
BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
^~~~~~~~~~~~~~~~~~~~~
SPI_CR1_LSBFIRST
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:42:43: error: 'SPI_BITORDER_MSBFIRST' was not declared in this scope
BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
^~~~~~~~~~~~~~~~~~~~~
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:42:43: note: suggested alternative: 'SPI_CR1_LSBFIRST'
BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
^~~~~~~~~~~~~~~~~~~~~
SPI_CR1_LSBFIRST
In file included from .pio\libdeps\STM32F103RC_btt_512K\Adafruit_MAX31865\Adafruit_MAX31865.cpp:17:0:
.pio\libdeps\STM32F103RC_btt_512K\Adafruit_MAX31865\Adafruit_MAX31865.h:84:32: warning: passing NULL to non-pointer argument 1 of 'Adafruit_SPIDevice::Adafruit_SPIDevice(int8_t, uint32_t, BitOrder, uint8_t, SPIClass*)' [-Wconversion-null]
Adafruit_SPIDevice spi_dev = NULL;
^~~~
In file included from c:\users\lastp.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\inttypes.h:22:0,
from C:\Users\lastp.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/avr/pgmspace.h:4,
from C:\Users\lastp.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/WString.h:29,
from C:\Users\lastp.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:47,
from C:\Users\lastp.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
from .pio\libdeps\STM32F103RC_btt_512K\Adafruit_MAX31865\Adafruit_MAX31865.h:50,
from .pio\libdeps\STM32F103RC_btt_512K\Adafruit_MAX31865\Adafruit_MAX31865.cpp:17:
.pio\libdeps\STM32F103RC_btt_512K\Adafruit_MAX31865\Adafruit_MAX31865.h:84:32: error: call to 'Adafruit_SPIDevice::Adafruit_SPIDevice(int8_t, uint32_t, BitOrder, uint8_t, SPIClass*)' uses the default argument for parameter 3, which is not yet defined
Adafruit_SPIDevice spi_dev = NULL;
^
.pio\libdeps\STM32F103RC_btt_512K\Adafruit_MAX31865\Adafruit_MAX31865.cpp: In constructor 'Adafruit_MAX31865::Adafruit_MAX31865(int8_t, int8_t, int8_t, int8_t)':
.pio\libdeps\STM32F103RC_btt_512K\Adafruit_MAX31865\Adafruit_MAX31865.cpp:38:76: error: call to 'Adafruit_SPIDevice::Adafruit_SPIDevice(int8_t, int8_t, int8_t, int8_t, uint32_t, BitOrder, uint8_t)' uses the default argument for parameter 6, which is not yet defined
spi_dev = Adafruit_SPIDevice(spi_cs, spi_clk, spi_miso, spi_mosi, 1000000);
^
.pio\libdeps\STM32F103RC_btt_512K\Adafruit_MAX31865\Adafruit_MAX31865.cpp: In constructor 'Adafruit_MAX31865::Adafruit_MAX31865(int8_t)':
.pio\libdeps\STM32F103RC_btt_512K\Adafruit_MAX31865\Adafruit_MAX31865.cpp:49:43: error: 'SPI_BITORDER_MSBFIRST' was not declared in this scope
Adafruit_SPIDevice(spi_cs, 1000000, SPI_BITORDER_MSBFIRST, SPI_MODE1);
^~~~~~~~~~~~~~~~~~~~~
.pio\libdeps\STM32F103RC_btt_512K\Adafruit_MAX31865\Adafruit_MAX31865.cpp:49:43: note: suggested alternative: 'SPI_CR1_LSBFIRST'
Adafruit_SPIDevice(spi_cs, 1000000, SPI_BITORDER_MSBFIRST, SPI_MODE1);
^~~~~~~~~~~~~~~~~~~~~
SPI_CR1_LSBFIRST
*** [.pio\build\STM32F103RC_btt_512K\lib1cd\Adafruit_MAX31865\Adafruit_MAX31865.cpp.o] Error 1
In file included from .pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214/Adafruit_BusIO_Register.h:2:0,
from .pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_BusIO_Register.cpp:1:
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:37:43: error: 'SPI_BITORDER_MSBFIRST' was not declared in this scope
BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
^~~~~~~~~~~~~~~~~~~~~
[... 27 lines cut ...]
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp: In member function 'void Adafruit_SPIDevice::transfer(uint8_t*, size_t)':
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp:99:31: error: no matching function for call to 'SPIClass::transfer(uint8_t*&, size_t&)'
_spi->transfer(buffer, len);
^
In file included from .pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:1:0,
from .pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp:1:
Marlin\src\HAL\STM32F1/SPI.h:268:11: note: candidate: uint8_t SPIClass::transfer(uint8_t) const
uint8_t transfer(uint8_t data) const;
^~~~~~~~
Marlin\src\HAL\STM32F1/SPI.h:268:11: note: candidate expects 1 argument, 2 provided
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp:115:23: error: 'SPI_BITORDER_LSBFIRST' was not declared in this scope
if (_dataOrder == SPI_BITORDER_LSBFIRST) {
^~~~~~~~~~~~~~~~~~~~~
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp:115:23: note: suggested alternative: 'SPI_CR1_LSBFIRST'
if (_dataOrder == SPI_BITORDER_LSBFIRST) {
^~~~~~~~~~~~~~~~~~~~~
SPI_CR1_LSBFIRST
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp:149:23: error: 'SPI_BITORDER_LSBFIRST' was not declared in this scope
if (_dataOrder == SPI_BITORDER_LSBFIRST) {
^~~~~~~~~~~~~~~~~~~~~
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp:149:23: note: suggested alternative: 'SPI_CR1_LSBFIRST'
if (_dataOrder == SPI_BITORDER_LSBFIRST) {
^~~~~~~~~~~~~~~~~~~~~
SPI_CR1_LSBFIRST
*** [.pio\build\STM32F103RC_btt_512K\lib2ab\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp.o] Error 1

ladyada, 2020-05-22:
ok moving the issue because its in the underlying i2c/spi support

caternuson, 2020-05-22:
Looks like a combo of that core not supporting `Wire.requestFrom(address, quantity, stop) `:
https://www.arduino.cc/en/Reference/WireRequestFrom

The other issue is none of the preprocessor `#define`s matching in `SPIDevice.h` where the typedefs are defined.

ladyada, 2020-05-22:
there's also no spi.transfer(buffer, size) for this platform

creatron, 2020-05-27:
Just change the following in platform ini file.. 

  ;Adafruit_MAX31865=https://github.com/adafruit/Adafruit_MAX31865/archive/master.zip
  Adafruit_MAX31865=https://github.com/adafruit/Adafruit_MAX31865/archive/1.1.0.zip
It should compile then   .. the bug was reported and bugfix should have it soon..

iamfraggle, 2020-05-29:
Is the upcoming bugfix a general one?  I have the same error messages trying to compile the MCP4728 library for a STM32F103C8.  As there is no version of the MCP4728 library that doesn't use BusIO, I don't have a downgrade workaround.

ladyada, 2020-05-29:
what is the instructions for installing the STM support in Arduino IDE? we could take a look if we could reproduce the error

iamfraggle, 2020-05-29:
https://github.com/stm32duino/wiki/wiki/Getting-Started

ladyada, 2020-05-30:
I just installed the STM package from stm32duino and compiled MCP4728 library just fine for STM32F103C8 in Arduino IDE 1.8.2 so this bug is not reproducable. please provide instructions on how to reproduce in arduino IDE and we'll fix it, we cannot until then

AzRockCrawler, 2020-06-07:
I believe I have a reproducible case which may or may not be of interest. The short version, the SPI library, from a specific board source listed below, does not provide an entry pointin SPI that matches the signature transfer(uint8_t*, size_t) expected by the Adafruit SPI wrapper.

It is possible there's a configuration error of some sort, but this is reproducible.

From fresh arduino install on Ubuntu 18.04
Add board manager from https://dan.drown.org/stm32duino/package_STM32duino_index.json 
Install STM32F1XX/GD32F1XX By STM32duino (version 2020.6.4)
Select Board: Generic STM32F103C Series, variant STM32F103C8 (20Kram. 64K flash)
Install ADAFRUIT_GFX library (v 1.8.4)
(allow above to install required Adafruit BUSIO v 1.3.2) 
Install MCUFriend_Kbv (v 2.9.8)

(Open a sketch that uses the touch screen)
#include <Adafruit_GFX.h>
#include "MCUFRIEND_kbv.h"

Compile 
Eror output:
Arduino: 1.8.12 (Linux), Board: "Generic STM32F103C series, STM32F103C8 (20k RAM. 64k Flash), STM32duino bootloader, 72Mhz (Normal), Smallest (default)"

/home/xxx/snap/arduino/current/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp: In member function 'void Adafruit_SPIDevice::transfer(uint8_t*, size_t)':
/home/xxx/snap/arduino/current/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp:99:31: error: no matching function for call to 'SPIClass::transfer(uint8_t*&, size_t&)'
     _spi->transfer(buffer, len);
                               ^
/home/xxx/snap/arduino/current/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp:99:31: note: candidate is:
In file included from /home/xxx/snap/arduino/current/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice.h:1:0,
                 from /home/xxx/snap/arduino/current/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp:1:
/home/xxx/snap/arduino/32/.arduino15/packages/stm32duino/hardware/STM32F1/2020.6.4/libraries/SPI/src/SPI.h:286:11: note: uint8 SPIClass::transfer(uint8) const
     uint8 transfer(uint8 data) const;
           ^
/home/xxx/snap/arduino/32/.arduino15/packages/stm32duino/hardware/STM32F1/2020.6.4/libraries/SPI/src/SPI.h:286:11: note:   candidate expects 1 argument, 2 provided
Multiple libraries were found for "Wire.h"
 Used: /home/xxx/snap/arduino/32/.arduino15/packages/stm32duino/hardware/STM32F1/2020.6.4/libraries/Wire
 Not used: /home/xxx/snap/arduino/32/.arduino15/packages/stm32duino/hardware/STM32F1/2020.6.4/libraries/WireSlave
exit status 1
Error compiling for board Generic STM32F103C series.
------------------
Looks like the SPI library from the contributed board source doesn't support calling transfer() with a length. Missing in the header and the implementation.

At line 96 in Adafruit_SPIDevie.cpp:
void Adafruit_SPIDevice::transfer(uint8_t *buffer, size_t len) {
  if (_spi) {
    // hardware SPI is easy
    _spi->transfer(buffer, len);
    return;
  }

This call at line 99 cannot be satisfied by the board's SPI library.

No regressions performed using older versions of the listed libraries.

ladyada, 2020-06-07:
why are you installing https://dan.drown.org/stm32duino/package_STM32duino_index.json and not https://github.com/stm32duino/Arduino_Core_STM32?

AzRockCrawler, 2020-06-07:
The reason I am doing it is because it was part of a set of instructions
for a specific project authored by someone else. 

Why the original author did it, I am unsure and can only speculate. I
can say that using the Core package doesn't compile with the author's
code, although I have many STM32 projects successfully using the Core
libraries. This is the first time trying the so called Blue Pill variant
which doesn't appear to be precisely matched in the Core board library.
I have not dug in to see why, but do see many version differences in a
number of libraries between the two board libraries. 

I'll try contacting the author to see what's up. I have seen a similar
issue reported in a few places. If there's anything actionable I'll
follow up. 

 

Links:
------
[1]
https://github.com/adafruit/Adafruit_BusIO/issues/27#issuecomment-640161253
[2]
https://github.com/

AzRockCrawler, 2020-06-07:
Looks like the Core libraries may not support the STM32F103C8T6 Timer2.
Again, cursory scan, have not dug in.

I am unsure why the dan.drown SPI hardware library does not support a
length parameter to the transfer() method as expected from the Core
library. 

 

Links:
------
[1]
https://github.com/adafruit/Adafruit_BusIO/issues/27#issuecomment-640161253
[2]
https://github.com/

AzRockCrawler, 2020-06-07:
Reverting to Adafruit_GFX 1.7.5 and dependent libraries allows the code
to compile with the dan.drown platform. 

Clearly there's a compatibility difference somewhere with the
STM32F103C8T6 variant. May dig in to identify it if I have the cycles. 

fwiw 

 

Links:
------
[1]
https://github.com/adafruit/Adafruit_BusIO/issues/27#issuecomment-640161253
[2]
https://github.com/

Cooltafel, 2020-09-30:
Hello, 

I got a problem with the BUSIO library. 
Recently, I got a waveshare 4.2" 3 color e-ink and a TTGO ESP32 Koala-T Wrover.
The example sketch of the GxEPD2 library isn't compiling due to an SPI error in the BUSIO library. 

Could someone help me? I searched everywhere, but no similar problem is reported (to my knowledge).

Error message:

`Build-opties gewijzigd, alles wordt opnieuw gebuild
...\Arduino\libraries\Adafruit_BusIO\Adafruit_SPIDevice.cpp: In member function 'void Adafruit_SPIDevice::transfer(uint8_t*, size_t)':

...\Arduino\libraries\Adafruit_BusIO\Adafruit_SPIDevice.cpp:119:31: error: no matching function for call to 'SPIClass::transfer(uint8_t*&, size_t&)'

     _spi->transfer(buffer, len);

                               ^

In file included from ...\Arduino\libraries\Adafruit_BusIO/Adafruit_SPIDevice.h:1:0,

                 from ...\Arduino\libraries\Adafruit_BusIO\Adafruit_SPIDevice.cpp:1:

...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\SPI\src/SPI.h:69:13: note: candidate: uint8_t SPIClass::transfer(uint8_t)

     uint8_t transfer(uint8_t data);

             ^

...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\SPI\src/SPI.h:69:13: note:   candidate expects 1 argument, 2 provided

exit status 1
Fout bij het compileren van board ESP32 Wrover Module
`

ladyada, 2020-09-30:
something is wrong with your setup, it absolutely works with ESP - update your board support

MadeInFck, 2020-12-12:
Hello, i've got exactly the same issue and my board support is up to date. 

/Users/mickael_fonck/Documents/Programmation/arduino/libraries/SPI/SPI.h:64:11: note: uint8_t SPIClass::transfer(uint8_t)
   uint8_t transfer(uint8_t data);
           ^
/Users/mickael_fonck/Documents/Programmation/arduino/libraries/SPI/SPI.h:64:11: note:   candidate expects 1 argument, 2 provided

Looks like only 1 argument is expected thought 2 are provided which is standard I guess: a pointer and number of bytes.

ladyada, 2020-12-12:
please supply a minimal example ... thousands of people use this library with ESP32 without issue...

MadeInFck, 2020-12-12:
Actually I'm using it with a wemos D1 mini and wemos TFT 1.4 v1.0.0:
#include <Adafruit_GFX.h> // Core graphics library
#include <Adafruit_ST7735.h> // Hardware-specific library
#include <SPI.h>
 
#define TFT_RST -1
#define TFT_CS D4
#define TFT_DC D3
 
Adafruit_ST7735 tft = Adafruit_ST7735(TFT_CS, TFT_DC, TFT_RST);
 
void setup(void)
{
tft.initR(INITR_144GREENTAB);
tft.setTextWrap(false); // Allow text to run off right edge
tft.fillScreen(ST7735_BLACK);
}
 
void loop(void)
{
tft.fillScreen(ST7735_BLACK);
tft.setCursor(0, 0);
tft.setTextColor(ST7735_BLUE);
tft.setTextSize(2);
tft.println("MADEINFCK.COM");
tft.setRotation(tft.getRotation() + 1);
delay(3000);
}

ladyada, 2020-12-12:
we were able to compile it just fine for ESP8266, please contact wemos for tech support - we are sure they have staff that can help their customers!

closed 2020-12-12 by ladyada

ahorn42, 2021-10-09:
I also had issues with that error message on a ESP8266. Updating the ESP8266 Board Definitions from a 2.x version to the 3.x versions with the Arduino board manager fixed it for me. @MadeInFck