adafruit/Adafruit_BusIO #28: Problem with LPC1769 compiling to use with HTS221 sensor.

astorun, 2020-05-27:
Hi there,

I am trying to add HTS221 sensor to BTT SKR 1.4 TURBO 3d printer board via SPI interface.https://www.amazon.com/BIGTREETECH-Controller-Compatible-With12864LCD-5TMC2209/dp/B082QYYFVX via Marlin Firmware.

This board has LPC1769 mcu. I added the necessary libraries to platformio but Adafruit-BusIO throws this error during compiling.

> In file included from lib\Adafruit_BusIO-master/Adafruit_BusIO_Register.h:2,
>                  from lib\Adafruit_HTS221-master/Adafruit_HTS221.h:22,
>                  from Marlin\src\lcd\dogm\status_screen_DOGM.cpp:44:
> lib\Adafruit_BusIO-master/Adafruit_SPIDevice.h:38:22: error: 'BitOrder' has not been declared
>    38 |                      BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
>       |                      ^~~~~~~~
> lib\Adafruit_BusIO-master/Adafruit_SPIDevice.h:43:22: error: 'BitOrder' has not been declared
>    43 |                      BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
>       |                      ^~~~~~~~
> lib\Adafruit_BusIO-master/Adafruit_SPIDevice.h:61:3: error: 'BitOrder' does not name a type
>    61 |   BitOrder _dataOrder;
>       |   ^~~~~~~~
> lib\Adafruit_BusIO-master/Adafruit_SPIDevice.h:38:43: error: 'SPI_BITORDER_MSBFIRST' was not declared in this scope
>    38 |                      BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
>       |                                           ^~~~~~~~~~~~~~~~~~~~~
> lib\Adafruit_BusIO-master/Adafruit_SPIDevice.h:39:41: error: 'SPI_MODE0' was not declared in this scope; did you mean 'SPI_MODE3'?
>    39 |                      uint8_t dataMode = SPI_MODE0, SPIClass *theSPI = &SPI);
>       |                                         ^~~~~~~~~
>       |                                         SPI_MODE3
> compilation terminated due to -fmax-errors=5.
> Compiling .pio\build\LPC1769\src\src\lcd\extui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\basic\commands.cpp.o
> *** [.pio\build\LPC1769\src\src\lcd\dogm\status_screen_DOGM.cpp.o] Error 1
> ======================================================================================= [FAILED] Took 56.65 seconds =======================================================================================
> 
> Environment               Status    Duration
> ------------------------  --------  ------------
> mega2560                  IGNORED
> mega1280                  IGNORED
> rambo                     IGNORED
> FYSETC_F6_13              IGNORED
> FYSETC_F6_14              IGNORED
> sanguino644p              IGNORED
> sanguino1284p             IGNORED
> melzi                     IGNORED
> melzi_optiboot            IGNORED
> at90usb1286_cdc           IGNORED
> at90usb1286_dfu           IGNORED
> DUE                       IGNORED
> DUE_USB                   IGNORED
> DUE_debug                 IGNORED
> LPC1768                   IGNORED
> LPC1769                   FAILED    00:00:56.648
> STM32F103RC               IGNORED
> STM32F103RC_fysetc        IGNORED
> STM32F103RC_btt           IGNORED
> STM32F103RC_btt_USB       IGNORED
> STM32F103RC_btt_512K      IGNORED
> STM32F103RC_btt_512K_USB  IGNORED
> STM32F103RE               IGNORED
> STM32F103RE_btt           IGNORED
> STM32F103RE_btt_USB       IGNORED
> STM32F4                   IGNORED
> STM32F7                   IGNORED
> ARMED                     IGNORED
> STM32F103VE_GTM32         IGNORED
> STM32F103VE_longer        IGNORED
> mks_robin_mini            IGNORED
> mks_robin_nano            IGNORED
> mks_robin                 IGNORED
> mks_robin_pro             IGNORED
> mks_robin_lite            IGNORED
> mks_robin_lite3           IGNORED
> jgaurora_a5s_a1           IGNORED
> STM32F103CB_malyan        IGNORED
> chitu_f103                IGNORED
> STM32F401VE_STEVAL        IGNORED
> FLYF407ZG                 IGNORED
> FYSETC_S6                 IGNORED
> STM32F407VE_black         IGNORED
> BIGTREE_SKR_PRO           IGNORED
> BIGTREE_GTR_V1_0          IGNORED
> BIGTREE_BTT002            IGNORED
> teensy31                  IGNORED
> teensy35                  IGNORED
> esp32                     IGNORED
> linux_native              IGNORED
> SAMD51_grandcentral_m4    IGNORED
> rumba32_f446ve            IGNORED
> rumba32_mks               IGNORED
> include_tree              IGNORED
> ================================================================================== 1 failed, 0 succeeded in 00:00:56.648 ================================================================================== 
> The terminal process terminated with exit code: 1
> 

Does it means LPC1769 not supported? If its a simple fix I would really like to use that sensor on my project.

I am not a programmer so I dont know how to solve this issue.

Thanks.

ladyada, 2020-05-27:
you need to add it here
https://github.com/adafruit/Adafruit_BusIO/blob/master/Adafruit_SPIDevice.h#L9
but i dont know what the define set for LPC1769 is, someone who is a programmer would have to tell us.

astorun, 2020-05-27:
Ok i think I fixed it for bitorder issue adding defined(ARDUINO_ARCH_LPC176X) to line 9 so it does not give the bitorder error anymore.

But the spimode error is still there and I dont know what it that means.

ladyada, 2020-05-27:
it means they did not make a full SPI implementation, is there any documentation of what MODEs are available? maybe they're called something else

astorun, 2020-05-27:
So I googled around and found out this page. https://www.exploreembedded.com/wiki/LPC1768:_SPI_Programming

PINSEL_FUNC_3 is displayed in bold. It probably means that its the SPI_MODE3.
Now I replaced it with SPI_MODE3. The error dissapears but now I have other errors in Adafruit_Sensor library.

lib\Adafruit_Sensor-master\Adafruit_Sensor.cpp: In member function 'void Adafruit_Sensor::printSensorDetails()':
lib\Adafruit_Sensor-master\Adafruit_Sensor.cpp:11:3: error: 'Serial' was not declared in this scope
   11 |   Serial.println(F("------------------------------------"));
      |   ^~~~Compiling .pio\build\LPC1769\lib98e\Adafruit_BusIO-master\Adafruit_SPIDevice.cpp.o
~~
lib\Adafruit_Sensor-master\Adafruit_Sensor.cpp:11:18: error: 'F' was not declared in this scope
   11 |   Serial.println(F("------------------------------------"));
      |                  ^
*** [.pio\build\LPC1769\libe72\Adafruit_Sensor-master\Adafruit_Sensor.cpp.o] Error 1
In file included from lib\Adafruit_BusIO-master/Adafruit_I2CDevice.h:1,
                 from lib\Adafruit_BusIO-master/Adafruit_BusIO_Register.h:1,
                 from lib\Adafruit_BusIO-master\Adafruit_BusIO_Register.cpp:1:
C:\Users\Ali\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Wire.h:53:13: error: 'size_t' does not name a type
   53 |     virtual size_t write(uint8_t);
      |             ^~~~~~
C:\Users\Ali\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Wire.h:26:1: note: 'size_t' is defined in header '<cstddef>'; did you forget to '#include <cstddef>'?
   25 | #include <inttypes.h>
  +++ |+#include <cstddef>
   26 |
C:\Users\Ali\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Wire.h:54:13: error: 'size_t' does not name a type
   54 |     virtual size_t write(const uint8_t *, size_t);
      |             ^~~~~~
C:\Users\Ali\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Wire.h:54:13: note: 'size_t' is defined in header '<cstddef>'; did you forget to '#include <cstddef>'?
C:\Users\Ali\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Wire.h:59:12: error: 'size_t' does not name a type
   59 |     inline size_t write(unsigned long n) { return write((uint8_t)n); }
      |            ^~~~~~
C:\Users\Ali\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Wire.h:59:12: note: 'size_t' is defined in header '<cstddef>'; did you forget to '#include <cstddef>'?
C:\Users\Ali\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Wire.h:60:12: error: 'size_t' does not name a type
   60 |     inline size_t write(long n) { return write((uint8_t)n); }
      |            ^~~~~~
C:\Users\Ali\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Wire.h:60:12: note: 'size_t' is defined in header '<cstddef>'; did you forget to '#include <cstddef>'?
C:\Users\Ali\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Wire.h:61:12: error: 'size_t' does not name a type
   61 |     inline size_t write(unsigned int n) { return write((uint8_t)n); }
      |            ^~~~~~
C:\Users\Ali\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Wire.h:61:12: note: 'size_t' is defined in header '<cstddef>'; did you forget to '#include <cstddef>'?
compilation terminated due to -fmax-errors=5.
In file included from lib\Adafruit_BusIO-master/Adafruit_I2CDevice.h:1,
                 from lib\Adafruit_BusIO-master\Adafruit_I2CDevice.cpp:1:
C:\Users\Ali\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Wire.h:53:13: error: 'size_t' does not name a type
   53 |     virtual size_t write(uint8_t);
      |             ^~~~~~
C:\Users\Ali\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Wire.h:26:1: note: 'size_t' is defined in header '<cstddef>'; did you forget to '#include <cstddef>'?
   25 | #include <inttypes.h>
  +++ |+#include <cstddef>
   26 |
C:\Users\Ali\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Wire.h:54:13: error: 'size_t' does not name a type
   54 |     virtual size_t write(const uint8_t *, size_t);
      |             ^~~~~~
C:\Users\Ali\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Wire.h:54:13: note: 'size_t' is defined in header '<cstddef>'; did you forget to '#include <cstddef>'?
C:\Users\Ali\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Wire.h:59:12: error: 'size_t' does not name a type
   59 |     inline size_t write(unsigned long n) { return write((uint8_t)n); }
      |            ^~~~~~
C:\Users\Ali\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Wire.h:59:12: note: 'size_t' is defined in header '<cstddef>'; did you forget to '#include <cstddef>'?
C:\Users\Ali\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Wire.h:60:12: error: 'size_t' does not name a type
   60 |     inline size_t write(long n) { return write((uint8_t)n); }
      |            ^~~~~~
C:\Users\Ali\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Wire.h:60:12: note: 'size_t' is defined in header '<cstddef>'; did you forget to '#include <cstddef>'?
C:\Users\Ali\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Wire.h:61:12: error: 'size_t' does not name a type
   61 |     inline size_t write(unsigned int n) { return write((uint8_t)n); }
      |            ^~~~~~
C:\Users\Ali\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Wire.h:61:12: note: 'size_t' is defined in header '<cstddef>'; did you forget to '#include <cstddef>'?
compilation terminated due to -fmax-errors=5.
[... cut ...]
esp32                     IGNORED
linux_native              IGNORED
SAMD51_grandcentral_m4    IGNORED
rumba32_f446ve            IGNORED
rumba32_mks               IGNORED
include_tree              IGNORED
================================================================================== 1 failed, 0 succeeded in 00:00:30.453 ================================================================================== 
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

ladyada, 2020-05-27:
hi there are some serious deficiencies in this Arduino implementation ... i dont know if it can be made to work without a lot of cleanup! you should ask the person who added this chip for help

astorun, 2020-05-27:
Ok I think I hit a deadend. Thanks for help anyways :) Time to try a DHT22.

ladyada, 2020-05-30:
yeah it does not look like SPI or Wire is supported by 
https://github.com/p3p/pio-framework-arduino-lpc176x
at all? if it isn't Arduino compatible, we dont guarantee it will work

closed 2020-05-30 by ladyada