adafruit/Adafruit_BusIO #29: Interrupts broken on the new attiny series since version 1.1.3

JasXSL, 2020-05-30:
Interrupts broke on the new attiny series (testing on attiny1604) when you started using BusIO. Using the following core: https://github.com/SpenceKonde/megaTinyCore

 First off if you try to compile it, you get the error:
```
Arduino\libraries\Adafruit_BusIO/Adafruit_SPIDevice.h:38:22: error: 'BitOrder' has not been declared
```

If you go in and remove the if statements, assuming it's an AVR board, it compiles but the interrupt pin is constantly held HIGH, never reacting to anything. It does this on all versions I've tried since then up to the latest arduino releases by today's date.

This sketch works as expected on 1.1.2, but for any version above it, the interrupt doesn't work:
https://gist.github.com/JasXSL/ed0f35498f64f8736e2fa730bde88b75

ladyada, 2020-05-30:
added support ifdefs will be released in 1.3.1 - dont' have hardware to test with but it does compile!

closed 2020-05-30 by ladyada

JasXSL, 2020-05-30:
Thanks :)

Using it with the LIS3DH library doesn't want to compile tho, both cloned from git just now. Want me to open an issue over there?

<pre>
\Arduino\libraries\Adafruit_LIS3DH\Adafruit_LIS3DH.cpp:107:52: error: no matching function for call to 'Adafruit_SPIDevice(int8_t&, long int, int, SPI_MODE_enum, SPIClass*&)'

                                        SPIinterface);
\Arduino\libraries\Adafruit_BusIO/Adafruit_SPIDevice.h:43:3: note: candidate: Adafruit_SPIDevice::Adafruit_SPIDevice(int8_t, int8_t, int8_t, int8_t, uint32_t, BitOrder, uint8_t) <near match>

   Adafruit_SPIDevice(int8_t cspin, int8_t sck, int8_t miso, int8_t mosi,

\Arduino\libraries\Adafruit_BusIO/Adafruit_SPIDevice.h:43:3: note:   conversion of argument 5 would be ill-formed:

Arduino\libraries\Adafruit_BusIO/Adafruit_SPIDevice.h:39:3: note: candidate: Adafruit_SPIDevice::Adafruit_SPIDevice(int8_t, uint32_t, BitOrder, uint8_t, SPIClass*) <near match>

   Adafruit_SPIDevice(int8_t cspin, uint32_t freq = 1000000,
</pre>

ladyada, 2020-05-30:
please post the screenshot of your tools menu

JasXSL, 2020-05-30:
[image]

ladyada, 2020-05-31:
ok try the latest push of busio

JasXSL, 2020-05-31:
It compiles! However, the interrupts are still broken, want me to make an issue on the LIS3DH repo for that?

ladyada, 2020-05-31:
yes - include the exact test program to reproduce the issue as you did here thanx!