adafruit/Adafruit_BusIO #30: Missing spi.h with Adafruit BusIO 1.3..2 in ESPhome

SeByDocKy, 2020-06-04:
Thank you for opening an issue on an Adafruit Arduino library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Arduino projects check these very common issues to ensure they don't apply**:

  - For uploading sketches or communicating with the board make sure you're using
    a **USB data cable** and **not** a **USB charge-only cable**.  It is sometimes
    very hard to tell the difference between a data and charge cable!  Try using the
    cable with other devices or swapping to another cable to confirm it is not
    the problem.

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and plug in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

  - **Ensure you are using an official Arduino or Adafruit board.** We can't
    guarantee a clone board will have the same functionality and work as expected
    with this code and don't support them.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Arduino board: ESP32 WROOM32

- Arduino IDE version (found in Arduino -> About Arduino menu):  platform: espressif32@1.12.1

- List the steps to reproduce the problem below : 

 Please find the error: https://pastebin.com/tmd26KJP

SeByDocKy, 2020-06-04:
> Looks like platformio?
> 
> > Arduino IDE version (found in Arduino -> About Arduino menu): platform: espressif32@1.12.1
> 
> ```
> In file included from /data/station_meteo/.piolibdeps/station_meteo/Adafruit BusIO_ID6214/Adafruit_SPIDevice.cpp:1:0:
> /data/station_meteo/.piolibdeps/station_meteo/Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:1:17: fatal error: SPI.h: No such file or directory
>  
> *************************************************************
> * Looking for SPI.h dependency? Check our library registry!
> *
> * CLI  > platformio lib search "header:SPI.h"
> * Web  > https://platformio.org/lib/search?query=header:SPI.h
> *
> *************************************************************
> ```

Yes was not sure ......... in fact since for me plateformia is completly transparent in backgroud of ESPhome ....

ladyada, 2020-06-04:
Yes SPI support is required for BusIO - ESP chips definitely have SPI support....so something is wrong with platformio's setup of ESPhome

closed 2020-06-04 by ladyada