adafruit/Adafruit_BusIO #37: Adafruit_I2CDevice write() const argument

mirzafahad, 2020-11-24:
This isn't an issue but more like a suggestion. Wouldn't it make more sense if **Adafruit_I2CDevice** _write()_ buffers are **const**?
The underlying TwoWire write()'s buffer argument is **const**, which **Adafruit_I2CDevice** uses. So, instead of:

`bool write(uint8_t *buffer, size_t len, bool stop, uint8_t *prefix_buffer, size_t prefix_len);`
`bool write_then_read(uint8_t *write_buffer, size_t write_len, uint8_t *read_buffer, size_t read_len, bool stop);`

it would make more sense if they are: 

`bool write(const uint8_t *buffer, size_t len, bool stop, const uint8_t *prefix_buffer, size_t prefix_len);`
`bool write_then_read(const uint8_t *write_buffer, ... ... ..., bool stop);`

Switching from **Wire.h** to **Adafruit_I2CDevice.h** breaks my code, just because the latter doesn't accept const.

ladyada, 2020-11-25:
sure, that would be ok but we have dozens of libraries that would also need to be updated. for now you could cast it.

mirzafahad, 2020-11-25:
I can do that. But if write() *buffer arguments are const, it will be able to take both const and non-const pointer without the cast. Any application that is already built based on these functions shouldn't be effected, as if nothing changed.

I have header files of constants pointing to sensor register addresses. Casting all these... 😭 

I am changing the local BUSIO library for now. Thanks 😃

ladyada, 2020-11-25:
ok you can submit a PR and we would try it!

mirzafahad, 2020-11-26:
Absolutely!

closed 2020-11-30 by mirzafahad