adafruit/Adafruit_BusIO #43: Compilation error when using megaTinyCore under PlatformIO

e-tinkers, 2021-03-12:
When using this library with megaTinyCore, I'm facing a compilation error under the  PlatformIO IDE, it is however doesn't seem to be an issue when compiling under Arduino IDE.

My Platformio.ini setting:

```
[env:ATtiny3217]
platform = atmelmegaavr
board = ATtiny3217
framework = arduino
lib_deps =
	adafruit/Adafruit SSD1306@^2.4.3
	adafruit/Adafruit MAX31856 library@^1.2.3
	br3ttb/PID@^1.2.1
	adafruit/Adafruit GFX Library@^1.10.6
```

The error shows: 
```
Processing ATtiny3217 (platform: atmelmegaavr; board: ATtiny3217; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelmegaavr/ATtiny3217.html
PLATFORM: Atmel megaAVR (1.4.0) > ATtiny3217
HARDWARE: ATTINY3217 16MHz, 2KB RAM, 32KB Flash
PACKAGES:
 - framework-arduino-megaavr-megatinycore 2.1.5
 - toolchain-atmelavr 2.70300.201015 (7.3.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 20 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Adafruit SSD1306> 2.4.3
|   |-- <Adafruit GFX Library> 1.10.6
|   |   |-- <Adafruit BusIO> 1.7.2
|   |   |   |-- <SPI> 1.0
|   |   |   |-- <Wire> 1.0
|   |   |-- <Wire> 1.0
|   |   |-- <SPI> 1.0
|   |-- <SPI> 1.0
|   |-- <Wire> 1.0
|-- <Adafruit MAX31856 library> 1.2.3
|   |-- <Adafruit BusIO> 1.7.2
|   |   |-- <SPI> 1.0
|   |   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <PID> 1.2.1
|-- <Adafruit GFX Library> 1.10.6
|   |-- <Adafruit BusIO> 1.7.2
|   |   |-- <SPI> 1.0
|   |   |-- <Wire> 1.0
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <EEPROM> 2.0
|-- <SPI> 1.0
|-- <Wire> 1.0
Building in release mode
Compiling .pio/build/ATtiny3217/src/main.cpp.o
Archiving .pio/build/ATtiny3217/lib5dd/libSPI.a
Indexing .pio/build/ATtiny3217/lib5dd/libSPI.a
Archiving .pio/build/ATtiny3217/lib024/libWire.a
In file included from .pio/libdeps/ATtiny3217/Adafruit MAX31856 library/Adafruit_MAX31856.h:102:
0,
                 from src/main.cpp:133:
.pio/libdeps/ATtiny3217/Adafruit BusIO/Adafruit_SPIDevice.h:62:22: error: 'BitOrder' has not bee
n declared
                      BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
                      ^~~~~~~~
.pio/libdeps/ATtiny3217/Adafruit BusIO/Adafruit_SPIDevice.h:67:22: error: 'BitOrder' has not bee
n declared
                      BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
                      ^~~~~~~~
.pio/libdeps/ATtiny3217/Adafruit BusIO/Adafruit_SPIDevice.h:88:3: error: 'BitOrder' does not nam
e a type
   BitOrder _dataOrder;
   ^~~~~~~~
Indexing .pio/build/ATtiny3217/lib024/libWire.a
Compiling .pio/build/ATtiny3217/libcf4/Adafruit BusIO/Adafruit_BusIO_Register.cpp.o
*** [.pio/build/ATtiny3217/src/main.cpp.o] Error 1
Compiling .pio/build/ATtiny3217/libcf4/Adafruit BusIO/Adafruit_I2CDevice.cpp.o
In file included from .pio/libdeps/ATtiny3217/Adafruit BusIO/Adafruit_BusIO_Register.h:2:0,
                 from .pio/libdeps/ATtiny3217/Adafruit BusIO/Adafruit_BusIO_Register.cpp:1:
.pio/libdeps/ATtiny3217/Adafruit BusIO/Adafruit_SPIDevice.h:62:22: error: 'BitOrder' has not bee
n declared
                      BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
                      ^~~~~~~~
.pio/libdeps/ATtiny3217/Adafruit BusIO/Adafruit_SPIDevice.h:67:22: error: 'BitOrder' has not bee
n declared
                      BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
                      ^~~~~~~~
.pio/libdeps/ATtiny3217/Adafruit BusIO/Adafruit_SPIDevice.h:88:3: error: 'BitOrder' does not nam
e a type
   BitOrder _dataOrder;
   ^~~~~~~~
*** [.pio/build/ATtiny3217/libcf4/Adafruit BusIO/Adafruit_BusIO_Register.cpp.o] Error 1
========================== [FAILED] Took 1.57 seconds ==========================
 ```

ladyada, 2021-03-12:
@SpenceKonde may have advice, we've never used that core :)

e-tinkers, 2021-03-13:
> @SpenceKonde may have advice, we've never used that core :)

I sorted of managed to make it work. 

If I look at the source code of [BusIO](https://github.com/adafruit/Adafruit_BusIO/blob/master/Adafruit_SPIDevice.h#L9), it does seems to cover the megaTinyCore. As I said, this works well on Arduino IDE, but failed on PlatformIO. 

If I change the line on the source code from `defined(MEGATINYCORE)` to `defined(MEGATINYCORE_MCU` after seeing the `Arduino.h` definition used in [megaTinyCore](https://github.com/SpenceKonde/megaTinyCore/blob/master/megaavr/cores/megatinycore/Arduino.h#L251-L253), the code compiled successfully for both PlatformIO and Arduino IDE now.

I think I will loop-in @ivankravets for his comment before deciding on how to fix this.

SpenceKonde, 2021-03-13:
Yikes! Thanks for letting me know about this, I'd carelessly removed that define, not considering that it might be relied upon to detect which core was in use. it will be corrected in the next release!

SpenceKonde, 2021-03-13:
!!! Waaaait.... the define is still in there - and yet if that fix is all it takes to correct this problem clearly indicates that those #defines aren't getting propagated...

SpenceKonde, 2021-03-13:
OHHH! It's PlatformIO, but not the Arduino IDE... (hadn't seen the start of the issue since came here from email) 

Ugh, this is some inconsistency between how Arduino and Platform IO behave regarding defines that are passed from platform.txt through the command line... 

I don't actually have platform IO set up (and have a list of higher priority items to work on a mile long), but I will put a band-aid for it into the next release - of course, it looks like it takes a while for releases to get propagated to PlatformIO :-(

e-tinkers, 2021-03-14:
> My recommendation for fixing it until this gets to platformIO would be adding to the Arduino.h for megaTinyCore something like:
> 
> ```
> #ifndef MEGATINYCORE
> #define MEGATINYCORE "2.1.5"
> #endif
> ```

Thanks for the recommendation @SpenceKonde. Now you mention the version, I realised that if you think it took a little time for PlatformIO to catch up the latest release, I'm surprise to see that the Arduino IDE is even behind, I noticed that the Arduino IDE is actually running version 2.0.5.
```
Using core 'arduino' from platform in folder: /Users/xxxx/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.0.5
```

and I now know what exactly caused this compilation error, as I can see that `MEGATINYCORE` is defined near the end of the `Arduino.h` on v2.0.5:
```
#define MEGATINYCORE "2.0.5"
```
This has been removed in v2.1.5 which is used on PlatformIO.

I will follow your recommendation for now to keep my project going. 

If I don't hear further comment from you, I will close this thread soon. Thanks for the help.

SpenceKonde, 2021-03-14:
Well, on the Arduino IDE you can go to tools -> board -> boards manager and use that to update it to the latest version. 2.2.8. 

Yeah - sometime in the 2.1.0-ish timeframe, I switched MEGATINYCORE from being a #define in Arduino.h to being a passed to the compiler as -DMEGATINYCORE=whatever hideous combination of backslashes and quotes it takes to get the define to make it through unscathed). That step was taken so that there could be a single source of truth regarding the version, previously it double or triple entry bookkeeping, and every source of a version number had a different number in it, and I think there was more than one release where none of them matched the actual version number it was released as..... I wonder if the stringify preprocessor directive could be used to do this more gracefully...

e-tinkers, 2021-03-14:
> I switched MEGATINYCORE from being a #define in Arduino.h to being a passed to the compiler as -DMEGATINYCORE=whatever hideous combination of backslashes and quotes it takes to get the define to make it through unscathed). 

Ah, thanks for the important piece of information, I add the custom build flags to the `platformio.ini`, and everything works perfectly now! My `platformio.ini` now looks like this:
```
[env:ATtiny3217]
platform = atmelmegaavr
board = ATtiny3217
framework = arduino
build_flags = -D MEGATINYCORE
lib_deps =
	adafruit/Adafruit SSD1306@^2.4.3
	adafruit/Adafruit MAX31856 library@^1.2.3
	br3ttb/PID@^1.2.1
	adafruit/Adafruit GFX Library@^1.10.6

Once again, thank you for your help.

closed 2021-03-14 by ladyada