adafruit/Adafruit_BusIO #44: Add check for I2C pull ups?

caternuson, 2021-04-01:
Any way we could protect against nothing being connected to the bus? This currently leads to code locking up. Here's a simple example:
```cpp
#include <Adafruit_I2CDevice.h>

#define ADDRESS 0x42

void setup() {
  Serial.begin(9600);
  while(!Serial);
  Serial.println("I2C TEST");

  Serial.println("Create.");
  Adafruit_I2CDevice *i2c_dev = new Adafruit_I2CDevice(ADDRESS, &Wire);

  Serial.println("Begin.");
  i2c_dev->begin();

  Serial.println("Write.");
  uint8_t buffer[1];
  buffer[0] = ADDRESS;
  i2c_dev->write(buffer, 1);

  Serial.println("Done.");
}

void loop() {
}
```

With pull ups connected, it gets through the code:
[image]

But **without** pull ups, it locks up:
[image]

This should really be something that Wire takes care of. But that's probably not a viable option.

Here's how CP does it for ref:
https://github.com/adafruit/circuitpython/blob/main/ports/atmel-samd/common-hal/busio/I2C.c#L79

ladyada, 2021-04-01:
oof
not really because we dont know what pins are being used for Wire...

caternuson, 2021-04-01:
:( good point....so probably not much that can be done, other than "if code locks up, check wiring".

This relates to this for more context:
https://github.com/adafruit/Adafruit_ADS1X15/issues/64

And just to illustrate the issue more lower level. Here's same thing just using Wire directly:
```cpp
#include <Wire.h>

#define ADDRESS 0x42

void setup() {
  Serial.begin(9600);
  while(!Serial);
  Serial.println("I2C TEST");

  Serial.println("Create.");
  TwoWire *theWire = &Wire;

  Serial.println("Begin.");
  theWire->begin();

  Serial.println("Write.");
  uint8_t buffer[1];
  buffer[0] = ADDRESS;
  theWire->beginTransmission(ADDRESS);
  theWire->write(buffer, 1);
  theWire->endTransmission();

  Serial.println("Done.");
}

void loop() {
}
```

**With pull ups:**
[image]

**Without pull ups:**
[image]

ladyada, 2021-04-01:
yeah basically - its part of why we added it to Circuit Python. in Arduino there's just no access to that information

caternuson, 2021-04-01:
ok, gonna close this then. good to at least have something to point to for when this comes up.

closed 2021-04-01 by caternuson