adafruit/Adafruit_BusIO #45: Compile error on my Arduino STM32 board, Adafruit-BMP085

rizacelik, 2021-05-12:
stm32f103c8t6  board.
I get the following error when compiling the code https://github.com/adafruit/Adafruit-BMP085-Library/blob/master/examples/BMP085test/BMP085test.ino at Arduino_STM32.

Arduino\libraries\Adafruit_BusIO\Adafruit_I2CDevice.cpp: In member function 'bool Adafruit_I2CDevice::write(const uint8_t*, size_t, bool, const uint8_t*, size_t)':
C:\Users\rizacelik\Documents\Arduino\libraries\Adafruit_BusIO\Adafruit_I2CDevice.cpp:88:47: error: invalid conversion from 'const uint8_t* {aka const unsigned char*}' to 'uint8* {aka unsigned char*}' [-fpermissive]
     if (_wire->write(prefix_buffer, prefix_len) != prefix_len) {

ladyada, 2021-05-12:
hi you need to ask the person who created teh STM32 board support package to fix their Wire->write arguments, it should be const!

closed 2021-05-12 by rizacelik

ilomon10, 2021-09-11:
Fix at [rogerclarkmelbourne/Arduino_STM32](https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/098a98c30e01eeb3f1cf9e9589a74a9263251e7c)