adafruit/Adafruit_BusIO #49: Chip Select assert missing from SPIDevice::transfer?

caternuson, 2021-06-19:
Wanted to use `transfer` for update of MCP3008 library to BusIO to facilitate the basic readout as a 3 byte xfer:
[image]

Was not getting any change in readings. Scoped it up and it would not even trigger. Trigger is based on CS line going low. So...took a look..and not seeing anything that is asserting CS in the `transfer` code similar to what is done in `write`, `read`, and `write_then_read`. Am I missing something?

ladyada, 2021-06-19:
correct, transfer() is same as SPI.transfer(), it's for non-CS SPI (e.g. like dotstars, shift registers or TLC chip)

caternuson, 2021-06-24:
Closing. This can be done, just needs a little extra work in user code. See example here:
https://github.com/adafruit/Adafruit_MCP3008/blob/6af029123b80a55b35eb3faeab40604b8eb0abf8/Adafruit_MCP3008.cpp#L101-L108

closed 2021-06-24 by caternuson