adafruit/Adafruit_BusIO #54: Problem with Adafruit_AHTX0 #53

vanderareosa, 2021-07-14:
trying to use Adafruit_AHTX0

Hardware:
Mac OS 10.13
Arduino IDE 1.8
Board ESP32-C3-DevKitM-1

I'm trying to upload an example to use the AHT10 temperature sensor but I'm getting the following error:

/Users/firemedia1/Library/Arduino15/packages/esp32/hardware/esp32/2.0.0-alpha1/cores/esp32/Arduino.h:107:66: note: suggested alternative: 'GPIO_IN_REG'
#define portInputRegister(port) ((volatile uint32_t*)((port)?GPIO_IN1_REG:GPIO_IN_REG))
^~~~~~~~~~~~
/Users/firemedia1/Dropbox/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp:58:33: note: in expansion of macro 'portInputRegister'
misoPort = (BusIO_PortReg *)portInputRegister(digitalPinToPort(misopin));
^~~~~~~~~~~~~~~~~

vanderareosa, 2021-07-14:
I had opened the topic in another location and only later opened it here because I found it more convenient. But I saw that it's already closed in the other place. Thanks.
I will wait for the fix since I depend on this library to make my project work.

ladyada, 2022-08-01:
pretty sure this is resolved

closed 2022-08-01 by ladyada