adafruit/Adafruit_BusIO #60: Unable to communicate with LIS3DH on MagTag via Arduino.

BeldrothTheGold, 2021-08-12:
- Arduino board:  **Adafruit MagTag 2.9"**

- Arduino IDE version (found in Arduino -> About Arduino menu):  **1.8.5**

- arduino-esp32 2.0.0 RC-1 installed via the boards manager

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **LIST REPRO STEPS BELOW**

Upload sketch, reboot and open Serial Monitor. 

I believe the issue is being caused by the esp32 implementation of the TwoWire::endTransmission() function expecting the user to call Wire.Write() prior to calling Wire.endTransmission()

`uint8_t TwoWire::endTransmission(bool sendStop)  // Assumes Wire.beginTransaction(), Wire.write()`
https://github.com/espressif/arduino-esp32/blob/2af8cc34850c445b1b0694653f086b9f9b1f52f9/libraries/Wire/src/Wire.cpp#L171

Minimum Sketch to reproduce issue:

```C
#include <Adafruit_LIS3DH.h>

Adafruit_LIS3DH lis = Adafruit_LIS3DH();

void setup() {  
  Serial.begin(115200);
  while (!Serial);
  
  if (! lis.begin(0x19)) {   // change this to 0x19 for alternative i2c address
    Serial.println("Couldnt start LIS3DH");
    while (1) yield();
  }

  Serial.print("Range = "); Serial.print(2 << lis.getRange());
  Serial.println("G");
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

[image]

This seems like an OK workaround:

Adding the following lines to the detected() function in Adafruit_I2CDevice.cpp
```c

bool Adafruit_I2CDevice::detected(void) {
  // Init I2C if not done yet
  if (!_begun && !begin()) {
    return false;
  }

  // A basic scanner, see if it ACK's
  _wire->beginTransmission(_addr);

#ifdef ESP32 
  // ESP32 implementation of wire expects write before endTransmission
  _wire->write(0);
#endif

  if (_wire->endTransmission() == 0) {

```

[image]

caternuson, 2021-08-12:
It could be this known issue with the ESP32-S2 RC1 BSP:
https://github.com/espressif/arduino-esp32/issues/5470

BeldrothTheGold, 2021-08-12:
> It could be this known issue with the ESP32-S2 RC1 BSP:
> [espressif/arduino-esp32#5470](https://github.com/espressif/arduino-esp32/issues/5470)

Indeed the PR, does fix this issue as well. OK. I will close my issue. Thanks!
https://github.com/espressif/arduino-esp32/commit/e5bd18d6aac65696483b56e8f33f9588f16c7b63

closed 2021-08-12 by BeldrothTheGold