adafruit/Adafruit_BusIO #61: Adafruit BusIO release version 1.9.0 breaks I2C reading of PM25AQI

zencow, 2021-08-21:
I just updated a bunch of libraries from Adafruit this evening, and I discovered reading PMSA003I devices over I2C began to fail after I reprogrammed a couple of my devices with the latest libraries.  I'm using an Adafruit Huzzah32 (esp32) with the Arduino programming environment.   The version of the Adafruit PM25AQI library is 1.0.6 and hasn't changed in about 6 months.  I investigated the Adafruit BusIO library as the possible culprit, since it was among the libraries I just updated and it had comments of "chunkify i2c" for the latest changes.

I did some quick testing by using the stock Example program for the Adafruit PM25AQI library, named "PM25_test", with both the latest version of Adafruit BusIO (1.9.0) and the previous version (1.8.3).

The output of "PM25_test" with BusIO version 1.8.3 is as expected.
The output of "PM25_test" with BusIO version 1.9.0 (after finding the PM25 sensor) is the repeating output: "Could not read from AQI"

This behavior is similar with my own custom program using these same libraries.

caternuson, 2021-08-21:
Just to make sure of the hardware setup, are you using these?
https://www.adafruit.com/product/4172
https://www.adafruit.com/product/4632
If not, can you link to product page(s) so we can test here as well.

zencow, 2021-08-21:
I'm using the feather version of the Huzzah32:
https://www.adafruit.com/product/3405

and yes, 4632 is the PMSA003I sensor I'm using:
https://www.adafruit.com/product/4632

The esp32 is plugged into a half-size breadboard using the QT connector to male header pins cable to connect to the sensor.  I currently have 2 of these in an identical setup.

-Chris

caternuson, 2021-08-21:
@zencow Thanks. Recreated issue and sent a potential fix with #62.

caternuson, 2021-08-22:
@zencow Please try the 1.9.1 release of BusIO:
https://github.com/adafruit/Adafruit_BusIO/releases/tag/1.9.1

zencow, 2021-08-22:
@caternuson The 1.9.1 release of BusIO seems to work fine with the PMSA003I sensor.  It also works fine with the SHT40 temp/humidity sensor I'm using in my project (that was not impacted by 1.9.0).  Thanks much!

caternuson, 2021-08-22:
Cool. Glad that worked. The 1.9.0 change was pretty significant. So still may be some bumps and edge cases out there. Let us know if you run into other weirdness. Thanks for finding this one.

closed 2021-08-22 by caternuson