adafruit/Adafruit_BusIO #63: I2C on different ports

hydrastarmaster, 2021-08-30:
In order to solve crowded projects where several sensors' I2C addresses overlap, it is possible to implement a bitbang transmission on different ports than the physical I2C (see e.g. library BitBang_I2C, which could be included as an extension). Please upgrade BusIO so as to enable all Adafruit sensor libraries to access this kind of communication protocol. This would allow connecting many sensors without collision and without physical multiplexers.

ladyada, 2021-08-30:
sure, if you can submit a non-breaking PR we'll take a look!

caternuson, 2021-08-30:
Could maybe do something like:
```cpp
Adafruit_I2CDevice::Adafruit_I2CDevice(uint8_t SCL_pin, uint8_t SDA_pin, uint8_t addr);
```
and add bit bang code to same class. But all sensor libraries would need to be updated also to actually utilize this.

Another idea might be to make a new class derived from `Wire` that overrides `read` and `write` with bit bang implementations. The pins could be passed in via ctor and/or a `setPins()` function. This could then just be passed in to existing sensor libraries.

hydrastarmaster, 2021-08-30:
The second is by far the more logical. It's the same mechanism as in the CS pin of SPI, or multiplexer channel selection. Comms problems are to be kept separate from sensors management (i.e. no impact on sensor libs), so it's surely better to override Wire.

ladyada, 2021-08-30:
@hydrastarmaster sounds like a plan. you don't even need to add code to this library, can have it be a separate library and then test it with Arduino in general. Closing since this is not dependant on BusIO anymore!

closed 2021-08-30 by ladyada