adafruit/Adafruit_BusIO #64: Chunks not working on ESP32

caternuson, 2021-09-24:
Related:
https://forums.adafruit.com/viewtopic.php?f=25&t=183433

Recreated using [pixels_test](https://github.com/adafruit/Adafruit_AMG88xx/blob/master/examples/pixels_test/pixels_test.ino) example from AMG88xx library.

[image]

Seems related to usage of `stop` parameter here:
https://github.com/adafruit/Adafruit_BusIO/blob/b903c7e0154cfe20ea618f972fad3277feae6fb2/Adafruit_I2CDevice.cpp#L179
which was not in previous chunk code:
https://github.com/adafruit/Adafruit_AMG88xx/blob/90e9eea8c43b6a173f10029029cec85ab1818823/Adafruit_AMG88xx.cpp#L234

ladyada, 2021-09-24:
erm, should probably pass `stop` thru there yeah?

caternuson, 2021-09-24:
It's passing through to the actual read start here:
https://github.com/adafruit/Adafruit_BusIO/blob/b903c7e0154cfe20ea618f972fad3277feae6fb2/Adafruit_I2CDevice.cpp#L193

But seems to be more fundamentally broken. Better example. Using a Feather ESP32. Nothing fancy, just try a read of 32 bytes:
```cpp
#include <Wire.h>

#define WIRE Wire
#define I2C_ADDRESS (0x69)
#define READ_SIZE (32)

void setup() {
  Serial.begin(9600);
  while(!Serial);

  uint8_t buffer[READ_SIZE];
  
  WIRE.begin();
  WIRE.requestFrom(I2C_ADDRESS, 32, true);
  uint32_t count = 0;
  while (WIRE.available() and count < READ_SIZE) {
    buffer[count++] = WIRE.read();
  }
}

void loop() {
}
```

Generates expected traffic:
[image]

Change:
```cpp
  WIRE.requestFrom(I2C_ADDRESS, 32, true);
```
to:
```cpp
  WIRE.requestFrom(I2C_ADDRESS, 32, false);
```
and nothing happens on the bus.

ladyada, 2021-09-24:
try using something else, esp 2.0.0 broke a ton of stuff

caternuson, 2021-09-24:
Just to document behavior on a different platform - **Qt Py M0**

**stop = true**
[image]

**stop = false**
[image]

ladyada, 2021-09-24:
k, looks right to me, there's a stop in the first one, none in the second

caternuson, 2021-09-24:
Yep. Just documenting expected behavior. Qt Py works as expected.

Just tried with ESP32 BSP 1.0.6 and same results as with BSP 2.0.0.

ladyada, 2021-09-24:
yah ok try 
https://github.com/espressif/arduino-esp32/issues/4729#issuecomment-922737664

caternuson, 2021-09-24:
hey! i know that pr :)
https://github.com/espressif/arduino-esp32/issues/5470#issuecomment-922738896

it does something different, but not the expected. basically same behavior either way.

**stop = true**
[image]

**stop = false**
[image]

ladyada, 2021-09-24:
ok please reply to that PR with the deets of what isnt workin

caternuson, 2022-08-22:
Closing. For now at least. The base issue for this is in the ESP32 BSP and, per comment here:
https://github.com/espressif/arduino-esp32/issues/5702#issuecomment-1111414087
can not be supported for ESP32.

closed 2022-08-22 by caternuson