adafruit/Adafruit_BusIO #69: error: 'class TwoWire' has no member named 'end'

alexqzd, 2021-11-25:
Our code stopped compiling today and broke our GitHub Actions workflow with the latest release. We are getting the following error:
```
.pio/libdeps/esp32dev/Adafruit BusIO@src-de43b9da31d281331630329764ceaa16/Adafruit_I2CDevice.cpp: In member function 'void Adafruit_I2CDevice::end()':
.pio/libdeps/esp32dev/Adafruit BusIO@src-de43b9da31d281331630329764ceaa16/Adafruit_I2CDevice.cpp:44:10: error: 'class TwoWire' has no member named 'end'
   _wire->end();
```

ESP32 with PlataformIO 
platformio/espressif32 @ 3.4.0

han-fastolfe, 2021-11-25:
I get this too. I was just about to report it, @alexqzd . ;-)

```
Processing wemos_d1_mini32 (platform: espressif32; board: wemos_d1_mini32; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/wemos_d1_mini32.html
PLATFORM: Espressif 32 (3.4.0) > WeMos D1 MINI ESP32
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10006.210326 (1.0.6) 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 42 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP32 HUB75 LED MATRIX PANEL DMA Display> 2.0.6
|   |-- <Adafruit GFX Library> 1.10.12
|   |   |-- <Adafruit BusIO> 1.9.7
|   |   |   |-- <Wire> 1.0.1
|   |   |   |-- <SPI> 1.0
|   |   |-- <Wire> 1.0.1
|   |   |-- <SPI> 1.0
|   |-- <FastLED> 3.4.0
|   |   |-- <SPI> 1.0
|-- <Adafruit BusIO> 1.9.7
|   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
|-- <Wire> 1.0.1
|-- <Adafruit GFX Library> 1.10.12
|   |-- <Adafruit BusIO> 1.9.7
|   |   |-- <Wire> 1.0.1
|   |   |-- <SPI> 1.0
|   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
|-- <SPI> 1.0
Building in release mode
Compiling .pio/build/wemos_d1_mini32/libdef/FastLED/FastLED.cpp.o
Compiling .pio/build/wemos_d1_mini32/libdef/FastLED/bitswap.cpp.o
Compiling .pio/build/wemos_d1_mini32/libdef/FastLED/colorpalettes.cpp.o
Compiling .pio/build/wemos_d1_mini32/libdef/FastLED/colorutils.cpp.o
Compiling .pio/build/wemos_d1_mini32/libdef/FastLED/hsv2rgb.cpp.o
Compiling .pio/build/wemos_d1_mini32/libdef/FastLED/lib8tion.cpp.o
Compiling .pio/build/wemos_d1_mini32/libdef/FastLED/noise.cpp.o
Compiling .pio/build/wemos_d1_mini32/libdef/FastLED/platforms.cpp.o
Compiling .pio/build/wemos_d1_mini32/libdef/FastLED/platforms/esp/32/clockless_rmt_esp32.cpp.o
Compiling .pio/build/wemos_d1_mini32/libdef/FastLED/power_mgt.cpp.o
Compiling .pio/build/wemos_d1_mini32/libdef/FastLED/wiring.cpp.o
Compiling .pio/build/wemos_d1_mini32/src/main.cpp.o
Generating partitions .pio/build/wemos_d1_mini32/partitions.bin
Compiling .pio/build/wemos_d1_mini32/lib2ea/Wire/Wire.cpp.o
Compiling .pio/build/wemos_d1_mini32/libcd9/SPI/SPI.cpp.o
Compiling .pio/build/wemos_d1_mini32/lib53d/Adafruit BusIO/Adafruit_BusIO_Register.cpp.o
Compiling .pio/build/wemos_d1_mini32/lib53d/Adafruit BusIO/Adafruit_I2CDevice.cpp.o
Compiling .pio/build/wemos_d1_mini32/lib53d/Adafruit BusIO/Adafruit_SPIDevice.cpp.o
Archiving .pio/build/wemos_d1_mini32/libcd9/libSPI.a
Indexing .pio/build/wemos_d1_mini32/libcd9/libSPI.a
/home/cory/Development/Arduino/sketchbook/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp: In static member function 'static void ESP32RMTController::interruptHandler(void*)':
/home/cory/Development/Arduino/sketchbook/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp:318:10: warning: unused variable 'stuff_to_do' [-Wunused-variable]
     bool stuff_to_do = false;
          ^
Compiling .pio/build/wemos_d1_mini32/lib4cd/Adafruit GFX Library/Adafruit_GFX.cpp.o
Compiling .pio/build/wemos_d1_mini32/lib4cd/Adafruit GFX Library/Adafruit_GrayOLED.cpp.o
Compiling .pio/build/wemos_d1_mini32/lib4cd/Adafruit GFX Library/Adafruit_SPITFT.cpp.o
Compiling .pio/build/wemos_d1_mini32/lib4cd/Adafruit GFX Library/glcdfont.c.o
.pio/libdeps/wemos_d1_mini32/Adafruit BusIO/Adafruit_I2CDevice.cpp: In member function 'void Adafruit_I2CDevice::end()':
.pio/libdeps/wemos_d1_mini32/Adafruit BusIO/Adafruit_I2CDevice.cpp:44:10: error: 'class TwoWire' has no member named 'end'
   _wire->end();
          ^
Compiling .pio/build/wemos_d1_mini32/lib5ba/ESP32 HUB75 LED MATRIX PANEL DMA Display/ESP32-HUB75-MatrixPanel-I2S-DMA.cpp.o
*** [.pio/build/wemos_d1_mini32/lib53d/Adafruit BusIO/Adafruit_I2CDevice.cpp.o] Error 1
========================================================== [FAILED] Took 2.28 seconds ==========================================================
The terminal process "platformio 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

The change in `Adafruit BusIO/Adafruit_I2CDevice.cpp` for 1.9.7 relies only on a `#define ESP8266` which is not there for ESP32.

ladyada, 2021-11-25:
ESP32 added Wire->end in 2.xx BSP
https://github.com/adafruit/Adafruit_BusIO/runs/4314549442?check_suite_focus=true#step:6:198
please try updating your ESP32 BSP! :)

han-fastolfe, 2021-11-25:
The regression is here. https://github.com/adafruit/Adafruit_BusIO/commit/6804a451b3a64cdd20f05dfb4544b4b6fe748571#diff-89c30e3923120a9c756032001dd2b9ddd1251a0d14bc669d570a9effa15baab1

alexqzd, 2021-11-25:
> ESP32 added Wire->end in 2.xx BSP
> https://github.com/adafruit/Adafruit_BusIO/runs/4314549442?check_suite_focus=true#step:6:198
> please try updating your ESP32 BSP! :)

PlatformIO says everything is already up-to-date
```
Platform Manager
================
Platform espressif32
--------
Updating platformio/espressif32               3.4.0                              [Up-to-date]
Updating platformio/toolchain-xtensa32        2.50200.97 @ ~2.50200.0            [Up-to-date]
Updating platformio/toolchain-esp32ulp        1.22851.191205 @ ~1.22851.0        [Up-to-date]
Updating platformio/framework-arduinoespressif32 3.10006.210326 @ ~3.10006.0        [Up-to-date]
Updating platformio/framework-espidf          3.40301.0 @ ~3.40301.0             [Up-to-date]
Updating platformio/tool-esptoolpy            1.30100.210531 @ ~1.30100.0        [Up-to-date]
Updating platformio/tool-mkspiffs             2.230.0 @ ~2.230.0                 [Up-to-date]
Updating platformio/tool-cmake                3.16.4 @ ~3.16.0                   [Up-to-date]
Updating platformio/tool-ninja                1.9.0 @ ^1.7.0                     [Up-to-date]
```

han-fastolfe, 2021-11-25:
Same here, all up to date.

```
pio platform update
cory@RedBarchetta[14:03:13]~/Documents/PlatformIO/Projects/ESP32Trinit Test 1$ pio platform update
Platform espressif32
--------
Updating platformio/espressif32               3.4.0                              [Up-to-date]
Updating platformio/toolchain-xtensa32        2.50200.97 @ ~2.50200.0            [Up-to-date]
Updating platformio/toolchain-esp32ulp        1.22851.191205 @ ~1.22851.0        [Up-to-date]
Updating espressif/toolchain-xtensa-esp32     8.4.0+2021r1 @ 8.4.0+2021r1        [Up-to-date]
Updating espressif/toolchain-xtensa-esp32s2   8.4.0+2021r1 @ 8.4.0+2021r1        [Up-to-date]
Updating espressif/toolchain-riscv32-esp      8.4.0+2021r1 @ 8.4.0+2021r1        [Up-to-date]
Updating platformio/framework-arduinoespressif32 3.10006.210326 @ ~3.10006.0        [Up-to-date]
Updating platformio/framework-espidf          3.40301.0 @ ~3.40301.0             [Up-to-date]
Updating platformio/tool-esptoolpy            1.30100.210531 @ ~1.30100.0        [Up-to-date]
Updating platformio/tool-mkspiffs             2.230.0 @ ~2.230.0                 [Up-to-date]
Updating platformio/tool-cmake                3.16.4 @ ~3.16.0                   [Up-to-date]
Updating platformio/tool-ninja                1.7.1 @ ^1.7.0                     [Up-to-date]
```

faster-light, 2021-11-25:
Plus one. Same error on ESP32

rbergen, 2021-11-25:
We're also running into this issue in [our project](https://github.com/PlummersSoftwareLLC/NightDriverStrip).

hathach, 2021-11-26:
can anyone tell me which arduino-esp core version you are running based on https://github.com/espressif/arduino-esp32/releases

hathach, 2021-11-26:
end() will be skipped for esp32 arch for now, we will enable it later on after a while to give time for user to update.

closed 2021-11-26 by hathach

rbergen, 2021-11-26:
I see this issue is already closed now (thank you very much for the fix!), but to answer your earlier question: the arduino-esp32 release would be 1.0.6. I have also attached a build log to this comment: [build.txt](https://github.com/adafruit/Adafruit_BusIO/files/7606874/build.txt).

hathach, 2021-11-26:
Wire::end() is added starting with core 2.0.1 . It is the currently latest, we decide to skip it for now to give user time to update. will re-enable it after a while

han-fastolfe, 2021-11-26:
> * can anyone tell me which arduino-esp core version you are running based on https://github.com/espressif/arduino-esp32/releases .
> * Also please attach your compile log as txt here (don't paste it since it is very long), I will try to use an macro specific to PIO/old version to opt-out the end()

That third component in the version number for `platformio/framework-arduinoespressif32` from what @alexqzd and I reported above,  `3.10006.210326`, looks an awful lot like a date. If it is, that means that `platformio/framework-arduinoespressif32` is still Espressif's 1.0.6 version and has yet to update to the 2.0.0 version which came out at the end of August and PIO has yet to update.

han-fastolfe, 2021-11-26:
@hathach Also, in the Boards manager for the Arduino IDE you can only install up to 1.0.6 for the Espressif32 Arduino stuff, neither 2.0.0 nor 2.0.1 are an option yet.

solhuebner, 2021-11-27:
Here I described a workaround for PlatformIO in case you want to use 2.0.1:

https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/issues/209#issuecomment-980062551

PR #70 (2021-11-26): Fix 32u with Wire end()
- use `WIRE_HAS_END` from arch AVR to detect whether wire has end(). This will work with both avr core that implement or not the wire end()
- Update: temporarily skip end() for esp32, which only implement end() starting from 2.0.1 (currently latest), this will give time for user to update their core (will enable later on). fix #69