adafruit/Adafruit_BusIO #82: Adafruit BusIO v1.11.1 breaks begin_SPI() on SAMD51 with BMP388

barry-ha, 2022-02-28:
I updated a bunch of libraries from Adafruit and I discovered that opening the BMP388 barometric pressure sensor over SPI is failing.  BusIO versions before v1.11.1 work fine.

Hardware
- https://www.adafruit.com/product/3857, Feather M4 Express, or
- https://www.adafruit.com/product/3800, ItsyBitsy M4 Express
- https://www.adafruit.com/product/3966, BMP388 barometric pressure sensor
- https://github.com/barry-ha/Griduino has my schematic in the 'hardware' folder

Software
- Arduino IDE 1.8.19 (or 1.8.16)
- https://github.com/adafruit/Adafruit_BMP3XX, using its example program 'bmp3xx_simpletest'
- changed two lines for ChipSelect and begin_SPI(), see code sketch below
- error message from sketch is "Could not find a valid BMP3 sensor, check wiring!"
- have tested a few Griduino assemblies, this hardware design has worked well for a year and is unchanged

Some recent BusIO changes include the comment "there doesn't seem to be a supported #define that this code can use to tell which core it's building against." This topic might be relevant to my SAMD51 devices. I don't know what the fix should be. For some other things, I've used:
- #if defined(SAMD_SERIES)
- #if defined(ADAFRUIT_ITSYBITSY_M4_EXPRESS)

Hope this helps.
Barry Hansen

## Code from bmp3xx_simpletest 
There are 2 changes for my wiring from Feather to BMP388. Look for "<---" markers.

    /***************************************************************************
      This is a library for the BMP3XX temperature & pressure sensor

      Designed specifically to work with the Adafruit BMP388 Breakout
      ----> http://www.adafruit.com/products/3966

      These sensors use I2C or SPI to communicate, 2 or 4 pins are required
      to interface.

      Adafruit invests time and resources providing this open source code,
      please support Adafruit and open-source hardware by purchasing products
      from Adafruit!

      Written by Limor Fried & Kevin Townsend for Adafruit Industries.
      BSD license, all text above must be included in any redistribution
     ***************************************************************************/

    #include <Wire.h>
    #include <SPI.h>
    #include <Adafruit_Sensor.h>
    #include "Adafruit_BMP3XX.h"

    #define BMP_SCK 13
    #define BMP_MISO 12
    #define BMP_MOSI 11
    #define BMP_CS 13     // <--- changed chip select

    #define SEALEVELPRESSURE_HPA (1013.25)

    Adafruit_BMP3XX bmp;

    void setup() {
      Serial.begin(115200);
      while (!Serial);
      Serial.println("Adafruit BMP388 / BMP390 test");

      if (! bmp.begin_SPI(BMP_CS)) {  // hardware SPI mode     <--- uncommented to use hardware SPI
        Serial.println("Could not find a valid BMP3 sensor, check wiring!");
        while (1);
      }

      // Set up oversampling and filter initialization
      bmp.setTemperatureOversampling(BMP3_OVERSAMPLING_8X);
      bmp.setPressureOversampling(BMP3_OVERSAMPLING_4X);
      bmp.setIIRFilterCoeff(BMP3_IIR_FILTER_COEFF_3);
      bmp.setOutputDataRate(BMP3_ODR_50_HZ);
    }

    void loop() {
      if (! bmp.performReading()) {
        Serial.println("Failed to perform reading :(");
        return;
      }
      Serial.print("Temperature = ");
      Serial.print(bmp.temperature);
      Serial.println(" *C");

      Serial.print("Pressure = ");
      Serial.print(bmp.pressure / 100.0);
      Serial.println(" hPa");

      Serial.print("Approx. Altitude = ");
      Serial.print(bmp.readAltitude(SEALEVELPRESSURE_HPA));
      Serial.println(" m");

      Serial.println();
      delay(2000);
    }

caternuson, 2022-02-28:
Can confirm. Running above example on an Itsy M4 and M0...

With Adafruit BusIO **v1.11.1**

Itsy M4
[image]

Itsy M0
[image]

With Adafruit BusIO **v1.11.0**

Itsy M4
[image]

Itsy M0
[image]

Seems most likely something with #77

ladyada, 2022-03-01:
@caternuson can you do a PR to revert #77 and see if that fixes it?

barry-ha, 2022-03-03:
Adafruit_BusIO v1.11.2 is approved. Thank you very much.

closed 2022-03-03 by barry-ha