adafruit/Adafruit_BusIO #86: Release 1.11.3 doesn't compile with Arduino 1.8.13

zencow, 2022-03-24:
- Arduino board:  **Adafruit ESP32 Feather**

- Arduino IDE version (found in Arduino -> About Arduino menu):  **1.8.13**

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **LIST REPRO STEPS BELOW**

I just updated this library from Adafruit (among others) and found it does not compile in Arduino or in PlatformIO.
my project has ```#include <Adafruit_LEDBackpack.h>``` which depends on ```Adafruit_BusIO```.

Downgrading to 1.8.12 allows me to compile and run my project.

In case it's relevant, I'm also still using ```arduino-esp32 1.0.4```.

I figured this would be adequate to locate the problem, but let me know if you want/need the verbose output.  Here are the errors from the Arduino IDE console output:

```
C:\Users\chris\Documents\Arduino\libraries\Adafruit_BusIO\Adafruit_SPIDevice.cpp: In member function 'bool Adafruit_SPIDevice::write(const uint8_t*, size_t, const uint8_t*, size_t)':
C:\Users\chris\Documents\Arduino\libraries\Adafruit_BusIO\Adafruit_SPIDevice.cpp:310:61: error: invalid conversion from 'const uint8_t* {aka const unsigned char*}' to 'uint8_t* {aka unsigned char*}' [-fpermissive]
       _spi->transferBytes(prefix_buffer, nullptr, prefix_len);
                                                             ^
In file included from C:\Users\chris\Documents\Arduino\libraries\Adafruit_BusIO\Adafruit_SPIDevice.h:9:0,
                 from C:\Users\chris\Documents\Arduino\libraries\Adafruit_BusIO\Adafruit_SPIDevice.cpp:1:
C:\Users\chris\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SPI\src/SPI.h:73:10: note:   initializing argument 1 of 'void SPIClass::transferBytes(uint8_t*, uint8_t*, uint32_t)'
     void transferBytes(uint8_t * data, uint8_t * out, uint32_t size);
          ^
C:\Users\chris\Documents\Arduino\libraries\Adafruit_BusIO\Adafruit_SPIDevice.cpp:313:47: error: invalid conversion from 'const uint8_t* {aka const unsigned char*}' to 'uint8_t* {aka unsigned char*}' [-fpermissive]
       _spi->transferBytes(buffer, nullptr, len);
                                               ^
In file included from C:\Users\chris\Documents\Arduino\libraries\Adafruit_BusIO\Adafruit_SPIDevice.h:9:0,
                 from C:\Users\chris\Documents\Arduino\libraries\Adafruit_BusIO\Adafruit_SPIDevice.cpp:1:
C:\Users\chris\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SPI\src/SPI.h:73:10: note:   initializing argument 1 of 'void SPIClass::transferBytes(uint8_t*, uint8_t*, uint32_t)'
     void transferBytes(uint8_t * data, uint8_t * out, uint32_t size);
          ^
C:\Users\chris\Documents\Arduino\libraries\Adafruit_BusIO\Adafruit_SPIDevice.cpp: In member function 'bool Adafruit_SPIDevice::write_then_read(const uint8_t*, size_t, uint8_t*, size_t, uint8_t)':
C:\Users\chris\Documents\Arduino\libraries\Adafruit_BusIO\Adafruit_SPIDevice.cpp:418:59: error: invalid conversion from 'const uint8_t* {aka const unsigned char*}' to 'uint8_t* {aka unsigned char*}' [-fpermissive]
       _spi->transferBytes(write_buffer, nullptr, write_len);
                                                           ^
In file included from C:\Users\chris\Documents\Arduino\libraries\Adafruit_BusIO\Adafruit_SPIDevice.h:9:0,
                 from C:\Users\chris\Documents\Arduino\libraries\Adafruit_BusIO\Adafruit_SPIDevice.cpp:1:
C:\Users\chris\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SPI\src/SPI.h:73:10: note:   initializing argument 1 of 'void SPIClass::transferBytes(uint8_t*, uint8_t*, uint32_t)'
     void transferBytes(uint8_t * data, uint8_t * out, uint32_t size);
          ^
```

yilungao, 2022-03-25:
Hi, have you tried upgrading arduino-esp32 to the latest version 2.0.2?

In 2.0.2 the esp32 SPI api has changed to the following:
`void transferBytes(const uint8_t * data, uint8_t * out, uint32_t size);`

ladyada, 2022-03-25:
we don't support esp32 v1 BSP - only the latest which passes CI fine!

closed 2022-03-25 by ladyada