adafruit/Adafruit_BusIO #100: Warning at Arduino Every Board

CombiesGit, 2022-06-29:
`E:\Programme\Arduino\portable\sketchbook\libraries\Adafruit_BusIO\Adafruit_I2CDevice.cpp: In member function 'bool Adafruit_I2CDevice::_read(uint8_t*, size_t, bool)':
E:\Programme\Arduino\portable\sketchbook\libraries\Adafruit_BusIO\Adafruit_I2CDevice.cpp:193:78: warning: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second:
  size_t recv = _wire->requestFrom((uint8_t)_addr, (uint8_t)len, (uint8_t)stop);
                                                                              ^
In file included from E:\Programme\Arduino\portable\sketchbook\libraries\Adafruit_BusIO\Adafruit_I2CDevice.h:5:0,
                 from E:\Programme\Arduino\portable\sketchbook\libraries\Adafruit_BusIO\Adafruit_I2CDevice.cpp:1:
E:\Programme\arduino\portable\packages\arduino\hardware\megaavr\1.8.7\libraries\Wire\src/Wire.h:64:12: note: candidate 1: size_t TwoWire::requestFrom(int, int, int)
     size_t requestFrom(int, int, int);
            ^~~~~~~~~~~
E:\Programme\arduino\portable\packages\arduino\hardware\megaavr\1.8.7\libraries\Wire\src/Wire.h:62:12: note: candidate 2: virtual size_t TwoWire::requestFrom(uint8_t, size_t, bool)
     size_t requestFrom(uint8_t, size_t, bool);
            ^~~~~~~~~~~
 `

That's a bit uncomfortable.
I don't see the reason for the cast here either

Here is a 'solution' that omits the cast on the AVR XMEGA.
 Adafruit_I2CDevice.cpp around line 191
`
// Schnipp
bool Adafruit_I2CDevice::_read(uint8_t *buffer, size_t len, bool stop) {
#if defined(TinyWireM_h)
  size_t recv = _wire->requestFrom((uint8_t)_addr, (uint8_t)len);
#elif defined(ARDUINO_ARCH_MEGAAVR)  
   size_t recv = _wire->requestFrom(_addr, len, stop);
#else
 size_t recv = _wire->requestFrom((uint8_t)_addr, (uint8_t)len, (uint8_t)stop);
#endif
// Schnapp
`

Are there any side effects I've overlooked?

hathach, 2022-07-25:
can you make a PR for suggested changes, it is easier to review changes in a PR.

ladyada, 2022-08-01:
i added, its totally untested. good luck!

closed 2022-08-01 by ladyada