adafruit/Adafruit_BusIO #103: i2cProcQueue >> IntegerDivideByZero

hdpklm, 2022-07-30:
Hi I use this library to control i2c devices via BLE to send commands from mobile to ESP32. My problem is that when I send many commands I get this error
`Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.`

So I dissasembly the code and find the error in this function `i2cProcQueue`.
I google this error and find that this error is due to unsafe thread crossing.

the driver or BLE is multithread, so I dont know how to use this code without  broke the multithread.

I think I have to lock the thread when I use the i2c.write and/or i2c.read. 
I think this option shuld by buildin the library

thank you

ladyada, 2022-08-01:
probably because you are using IRQs? I2C locking/unlocking is not done by this library because its not standard arduino. however, if you have a way to add it without breaking other boards, please submit a PR! 
closing now because this is not something we are planning on adding ourselves

closed 2022-08-01 by ladyada

hdpklm, 2023-02-08:
Finaly I find the problem.
you cant resend Start Command before Stop. you have to stop old request and then send the new request