adafruit/Adafruit_BusIO #109: Pointer to hardware SPI not correctly initialized if hardware is available but software SPI is used

Ezekiel-DA, 2022-11-01:
In `Adafruit_SPIDevice.h`:

```
class Adafruit_SPIDevice {
<snip>
#ifdef BUSIO_HAS_HW_SPI
  SPIClass *_spi;
  SPISettings *_spiSetting;
#else
  uint8_t *_spi = nullptr;
  uint8_t *_spiSetting = nullptr;
#endif
```

If hardware SPI is available, `_spi` is not initialized to `nullptr`.

If you then use the software SPI constructor, `_spi` remains unintialized.

Then, in `Adafruit_SPIDevice.cpp`:
```
bool Adafruit_SPIDevice::begin(void) {
  if (_cs != -1) {
    pinMode(_cs, OUTPUT);
    digitalWrite(_cs, HIGH);
  }

  if (_spi) { // hardware SPI
#ifdef BUSIO_HAS_HW_SPI
    _spi->begin();
#endif
```

`if(_spi)` is used to detect hardware or software SPI mode; since the value of `_spi` is random and possibly non zero, `_spi->begin()` is called and promptly causes crashes / undefined behavior.

Ezekiel-DA, 2022-11-02:
This fixes the crash I was seeing, predictably!

Thanks for looking into it!

caternuson, 2022-11-02:
Cool. Thanks for testing and verifying. That helped. The change has been merged and released:
https://github.com/adafruit/Adafruit_BusIO/releases/tag/1.14.1
Can update library to that once it shows up in Arduino Library Manager.

closed 2022-11-07 by Ezekiel-DA