adafruit/Adafruit_BusIO #111: If NO_GLOBAL_INSTANCES or NO_GLOBAL_SERIAL is defined, not able to find Serial in Adafruit_BusIO_Register.h

fionn-r, 2023-02-01:
- Arduino board:  ESP32 WROOM32 module

- Arduino IDE version (found in Arduino -> About Arduino menu):  Using VS Code with ESPIDF, CMake, Arduino as a submodule: espressif/arduino-esp32 v2.0.5

Compile with `-DNO_GLOBAL_SERIAL`. Causes HardwareSerial.h to not `extern HardwareSerial Serial;` which means that [Adafruit_BusIO_Register.h #L74](https://github.com/adafruit/Adafruit_BusIO/blob/186bfb005c6d4b597a09290cb3e451c30a5baa97/Adafruit_BusIO_Register.h#L74) tries to use &Serial as a default argument but it doesn't exist.

Solution would be to use the same define guard around the method declarations so if NO_GLOBAL_INSTANCES or NO_GLOBAL_SERIAL is defined, it will not add in a default argument.

fionn-r, 2023-02-01:
I've ended up making a pull request since it's a super easy fix

caternuson, 2023-02-01:
It sounds like you are potentially trying to use this as a general purpose library for a generic target board. Can you recreate this via the Arduino IDE? Or point to a specific board in the ESP32 Arduino BSP that would cause this?

fionn-r, 2023-02-01:
Hi, yes it is on a custom board where for various reasons the two serials are reversed but it is using the ESP32 module which is, I guess, just the one called ESP32?

I would say it's also not an unusual thing to want to redefine the Serial ports and is possible on all boards

caternuson, 2023-02-01:
For updating this library, the issue would need to be recreated for a target board in the ESP32 BSP via the Arduino IDE. One that shows up in the Tools>Board list in the IDE. Not aware of one that does not define `Serial`. 

If this won't work for your custom board, you can fork this library and change as you wish in the fork.

fionn-r, 2023-02-02:
To clarify, to recreate the problem with any board, all you have to do is compile with `-DNO_GLOBAL_SERIAL` as this is something that is baked into HardwareSerial. But I do understand if you don't want to include it, just since it is a built in feature of Arduino,  guess it makes sense to support all features of Arduino.

caternuson, 2023-02-02:
Yep, these are Arduino libraries, with the intention of supporting all Arduino platforms to the extent possible for use in the Arduino IDE.

closed 2023-02-02 by caternuson