adafruit/Adafruit_BusIO #128: DEBUG_SERIAL decimal address printed as hex

disq, 2024-04-29:
```
#ifdef DEBUG_SERIAL
  DEBUG_SERIAL.print(F("Address 0x"));
  DEBUG_SERIAL.print(_addr);
#endif
```

`_addr` is printed as decimal, so for example a failing LIS3DH init shows as "Address 0x24 Not detected" where it's actually `0x18` (or, just `24`)

https://github.com/adafruit/Adafruit_BusIO/blob/1.16.0/Adafruit_I2CDevice.cpp#L72

caternuson, 2024-04-29:
Good catch. It needs a `HEX` parameter like other similar debug prints, ex:
https://github.com/adafruit/Adafruit_BusIO/blob/e1aa388418ef4d2a326a58dd62620fb954799ccf/Adafruit_I2CDevice.cpp#L135

Want to PR a fix? No worries if you don't. But this would make for a good first pull request.

ladyada, 2025-01-08:
fixed recently

closed 2025-01-08 by ladyada