adafruit/Adafruit_BusIO #133: SPI.h Not Found when the sensor I'm using doesn't even use SPI? Apple Silicon on PlatformIO 6.1.16/VSCode using adafruit/Adafruit BusIO@^1.14.1

CodeBradley, 2024-09-30:
I'm using a FLIP C3 board, which does not have SPI pins (only I2C, UART, digital, analog). I've tried declaring an SPI pinout in the pins_arduino.h file just to see if it fixes the error, but that didn't work. Suggestions online say to include SPI.h in the beginning of my script, but the same error occurs. It also continues happening when setting it in `lib_deps`.

What's the official way of resolving this?

platformio.ini
```
[platformio]
default_envs = flip_c3

[env:flip_c3]
platform = espressif32
board = flip_c3
framework = arduino
build_flags = -I /variants/flip_c3
lib_deps =
	adafruit/Adafruit NeoPixel@^1.12.3
	adafruit/Adafruit TCS34725@^1.4.0
    adafruit/Adafruit BusIO@^1.14.1

[env:seeed_xiao_esp32s3]
platform = espressif32
board = seeed_xiao_esp32s3
framework = arduino
platform_packages =
    framework-arduinoespressif32
lib_deps = 
	SPI
	adafruit/Adafruit NeoPixel@^1.12.3
	adafruit/Adafruit TCS34725@^1.4.0
    adafruit/Adafruit BusIO@^1.14.1
```

main.cpp
```
#include <Wire.h>
#include "Adafruit_TCS34725.h"
#include <Adafruit_NeoPixel.h>

Adafruit_NeoPixel strip = Adafruit_NeoPixel(16, GPIO6, NEO_GRB + NEO_KHZ800);
Adafruit_TCS34725 tcs = Adafruit_TCS34725(TCS34725_INTEGRATIONTIME_614MS, TCS34725_GAIN_1X);

// IMPORTANT: To reduce NeoPixel burnout risk, add 1000 uF capacitor across
// pixel power leads, add 300 - 500 Ohm resistor on first pixel's data input
// and minimize distance between Arduino and first pixel.  Avoid connecting
// on a live circuit...if you must, connect GND first.

void setup() {

  Serial.begin(115200);
  while(!Serial);
  if (tcs.begin()) {
    Serial.println("Found sensor");
  } else {
    Serial.println("No TCS34725 found ... check your connections");
    while (1);
  }
  strip.setBrightness(50);
  strip.begin();
  strip.show(); // Initialize all pixels to 'off'
  
}

void loop() {

  uint16_t r, g, b, c, colorTemp, lux;
  tcs.getRawData(&r, &g, &b, &c);
  // colorTemp = tcs.calculateColorTemperature(r, g, b);
  colorTemp = tcs.calculateColorTemperature_dn40(r, g, b, c);
  lux = tcs.calculateLux(r, g, b);
  Serial.print("Color Temp: "); Serial.print(colorTemp, DEC); Serial.print(" K - ");
  Serial.print("Lux: "); Serial.print(lux, DEC); Serial.print(" - ");
  Serial.print("R: "); Serial.print(r, DEC); Serial.print(" ");
  Serial.print("G: "); Serial.print(g, DEC); Serial.print(" ");
  Serial.print("B: "); Serial.print(b, DEC); Serial.print(" ");
  Serial.print("C: "); Serial.print(c, DEC); Serial.print(" ");
  Serial.println(" ");

}

// Fill the dots one after the other with a color
void colorWipe(uint32_t c, uint8_t wait) {
  for(uint16_t i=0; i<strip.numPixels(); i++) {
      strip.setPixelColor(i, c);
      strip.show();
      delay(wait);
  }
}
```

Error:
```
Compiling .pio/build/flip_c3/FrameworkArduino/HWCDC.cpp.o
Compiling .pio/build/flip_c3/FrameworkArduino/HardwareSerial.cpp.o
In file included from .pio/libdeps/flip_c3/Adafruit BusIO/Adafruit_BusIO_Register.h:10,
                 from .pio/libdeps/flip_c3/Adafruit BusIO/Adafruit_BusIO_Register.cpp:1:
.pio/libdeps/flip_c3/Adafruit BusIO/Adafruit_SPIDevice.h:9:10: fatal error: SPI.h: No such file or directory

*************************************************************
* Looking for SPI.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SPI.h"
* Web  > https://registry.platformio.org/search?q=header:SPI.h
*
*************************************************************

 #include <SPI.h>
          ^~~~~~~
compilation terminated.
*** [.pio/build/flip_c3/libf72/Adafruit BusIO/Adafruit_BusIO_Register.cpp.o] Error 1
In file included from .pio/libdeps/flip_c3/Adafruit BusIO/Adafruit_SPIDevice.cpp:1:
.pio/libdeps/flip_c3/Adafruit BusIO/Adafruit_SPIDevice.h:9:10: fatal error: SPI.h: No such file or directory
```

There's claims that the problem is caused by how PIO scans the code, but the workaround does not help if the board I'm using doesn't have SPI.
https://community.platformio.org/t/adafruit-busio-adafruit-spidevice-h17-fatal-error-spi-h-no-such-file-or-directory/14864/10

What's the official way of resolving this?

caternuson, 2024-09-30:
Related to #127, but in this case SPI instead of Serial.

Not aware of a possible work around for platformio. Is there anyway to delete specific files from a library? Like deleting `Adafruit_BusIO_Register.h` and `Adafruit_BusIO_Register.cpp`?

CodeBradley, 2024-10-02:
I got it to compile on the board I was having issues with, but not sure of an official way to fix it since I'm not even using SPI.

For anyone else having problems, I created this PIO board myself and the docs did not mention it having SPI so I left it out of the `variants/pins_arduino.h` file. Adafruit complained about SPI which forced me to import it (I did it in lib_deps, but an include in main.cpp should work too). Once I did this I ran into this error during compilation:

```
~/.platformio/packages/framework-arduinoespressif32/libraries/SPI/src/SPI.cpp: In member function 'void SPIClass::begin(int8_t, int8_t, int8_t, int8_t)':
~/.platformio/packages/framework-arduinoespressif32/libraries/SPI/src/SPI.cpp:93:16: error: 'SCK' was not declared in this scope
         _sck = SCK;
                ^~~
~/.platformio/packages/framework-arduinoespressif32/libraries/SPI/src/SPI.cpp:93:16: note: suggested alternative: 'SCL'
         _sck = SCK;
                ^~~
                SCL
~/.platformio/packages/framework-arduinoespressif32/libraries/SPI/src/SPI.cpp:94:17: error: 'MISO' was not declared in this scope
         _miso = MISO;
                 ^~~~
~/.platformio/packages/framework-arduinoespressif32/libraries/SPI/src/SPI.cpp:94:17: note: suggested alternative: 'EIO'
         _miso = MISO;
                 ^~~~
                 EIO
~/.platformio/packages/framework-arduinoespressif32/libraries/SPI/src/SPI.cpp:95:17: error: 'MOSI' was not declared in this scope
         _mosi = MOSI;
                 ^~~~
~/.platformio/packages/framework-arduinoespressif32/libraries/SPI/src/SPI.cpp:95:17: note: suggested alternative: 'M_PI'
         _mosi = MOSI;
                 ^~~~
                 M_PI
~/.platformio/packages/framework-arduinoespressif32/libraries/SPI/src/SPI.cpp:96:15: error: 'SS' was not declared in this scope
         _ss = SS;
               ^~
*** [.pio/build/flip_c3/libf61/SPI/SPI.cpp.o] Error 1
```

For anyone else running into this problem, odds are you are using a board that doesn't have any of the SPI pins defined in it's variants file. If you're using them then obviously you should add the correct pins to the `variants/pins_arduino.h` file, or to the main.cpp file. If the board doesn't have it or you're not using SPI then you can probably just define random pin numbers in the variants or main.cpp file and it should resolve this issue.

---
### VDBX Flip C3
**~/.platformio/platforms/espressif32/boards/flip_c3.json**
```
{
  "build": {
    "arduino": {
      "ldscript": "esp32c3_out.ld"
    },
    "core": "esp32",
    "extra_flags": "-DARDUINO_FLIP_C3",
    "f_cpu": "160000000L",
    "f_flash": "40000000L",
    "flash_mode": "dio",
    "mcu": "esp32c3",
    "variant": "flip_c3"
  },
  "connectivity": [
    "bluetooth",
    "wifi"
  ],
  "debug": {
    "default_tool": "esp-builtin",
    "openocd_target": "esp32c3",
    "tools": {
      "esp-builtin": {
        "onboard": true,
        "require_debug_port": true,
        "default": true
      }
    }
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "VDBX Flip C3",
  "upload": {
    "flash_size": "4MB",
    "maximum_ram_size": 327680,
    "maximum_size": 4194304,
    "protocol": "esptool",
    "require_upload_port": true,
    "speed": 460800
  },
  "url": "https://wiki.vdbx.io/products/flip_c3",
  "vendor": "VDBX"
}
```

**~/.platformio/packages/framework-arduinoespressif32/variants/flip_c3/pins_arduino.h**
```
#ifndef Pins_Arduino_h
#define Pins_Arduino_h

#include <stdint.h>
#include <stdbool.h>

// Define the number of pins
#define EXTERNAL_NUM_INTERRUPTS 10
#define NUM_DIGITAL_PINS        10
#define NUM_ANALOG_INPUTS       10

static const uint8_t LED_BUILTIN = 10;
#define BUILTIN_LED  LED_BUILTIN // backward compatibility

// I2C
static const uint8_t SCL = 0;
static const uint8_t SDA = 1;

// Pin definitions based on Flip C3 schematic
static const uint8_t GPIO0  = 0;
static const uint8_t GPIO1  = 1;
static const uint8_t GPIO2  = 2;
static const uint8_t GPIO3  = 3;
static const uint8_t GPIO4  = 4;
static const uint8_t GPIO5  = 5;
static const uint8_t GPIO6  = 6;
static const uint8_t GPIO7  = 7;
static const uint8_t GPIO8  = 8; //WS2812B RGBOUT
static const uint8_t GPIO9  = 9; //Boot button

// UART
static const uint8_t RX = 20;
static const uint8_t TX = 21;

// Analog pins
static const uint8_t A1 = GPIO2;
static const uint8_t A2 = GPIO3;
static const uint8_t A3 = GPIO4;
static const uint8_t A5 = GPIO5;
static const uint8_t A6 = GPIO6;
static const uint8_t A7 = GPIO7;
static const uint8_t A8 = GPIO8;
static const uint8_t A9 = GPIO9;

#endif /* Pins_Arduino_h */
```