adafruit/Adafruit_BusIO #141: Guru meditation error on 1.17, but not on 1.16

bolkedebruin, 2025-02-09:
While using the older, retired, MCP23008 driver but also the newer MCP23017 driver I get the following stack trace on 1.17. With the older driver this happens immediately with the newer it happens less often but conistently.

```
0x421675e4: Adafruit_GenericDevice::readRegister(unsigned char*, unsigned char, unsigned char*, unsigned short) at [REDACTED]/Adafruit BusIO/Adafruit_GenericDevice.cpp:60
0x4213a0a2: Adafruit_BusIO_Register::read(unsigned char*, unsigned char) at [REDACTED]/Adafruit BusIO/Adafruit_BusIO_Register.cpp:250
0x4213a0b9: Adafruit_BusIO_Register::read() at [REDACTED]/Adafruit BusIO/Adafruit_BusIO_Register.cpp:189
0x42046e4b: Adafruit_MCP23008::read8(unsigned char) at [REDACTED]/Adafruit MCP23008 library/Adafruit_MCP23008.cpp:177
0x42046e5d: Adafruit_MCP23008::readGPIO() at [REDACTED]/Adafruit MCP23008 library/Adafruit_MCP23008.cpp:98
```

The offending line is ```return _genericdevice->readRegister(addrbuffer, _addrwidth, buffer, len);``` in the function below. The device is set as a I2C device some I am not quite sure why the generic function is called. For now I am reverting to the older version.

```c
bool Adafruit_BusIO_Register::read(uint8_t *buffer, uint8_t len) {
  uint8_t addrbuffer[2] = {(uint8_t)(_address & 0xFF),
                           (uint8_t)(_address >> 8)};
  if (_i2cdevice) {
    return _i2cdevice->write_then_read(addrbuffer, _addrwidth, buffer, len);
  }
  if (_spidevice) {
    if (_spiregtype == ADDRESSED_OPCODE_BIT0_LOW_TO_WRITE) {
      // very special case!
      // pass the special opcode address which we set as the high byte of the
      // regaddr
      addrbuffer[0] =
          (uint8_t)(_address >> 8) | 0x01; // set bottom bit high to read
      // the 'actual' reg addr is the second byte then
      addrbuffer[1] = (uint8_t)(_address & 0xFF);
      // the address appears to be a byte longer
      return _spidevice->write_then_read(addrbuffer, _addrwidth + 1, buffer,
                                         len);
    }
    if (_spiregtype == ADDRBIT8_HIGH_TOREAD) {
      addrbuffer[0] |= 0x80;
    }
    if (_spiregtype == ADDRBIT8_HIGH_TOWRITE) {
      addrbuffer[0] &= ~0x80;
    }
    if (_spiregtype == AD8_HIGH_TOREAD_AD7_HIGH_TOINC) {
      addrbuffer[0] |= 0x80 | 0x40;
    }
    return _spidevice->write_then_read(addrbuffer, _addrwidth, buffer, len);
  }
  if (_genericdevice) {
    return _genericdevice->readRegister(addrbuffer, _addrwidth, buffer, len);
  }
  return false;
}
```

caternuson, 2025-02-10:
Please provide an example sketch do demonstrate the issue.

steinundfloete, 2025-03-26:
I have a similar problem with 1.17.0, see stack trace below
This happens on calling mcp.digitalWrite(4, 0);
But that was successfully called before multiple times.
I was setting all ports on the MCP23008 to OUTPUT and switching 6 LEDs with a 1.2k resistor to GND each.
Usually I use mcp.writeGPIO(x), but for fast single LED updates (max 20 per second), I use digitalWrite

`
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x401757c9  PS      : 0x00060730  A0      : 0x800d6180  A1      : 0x3ffb2120  
A2      : 0x0000003c  A3      : 0x3ffb214e  A4      : 0x00000001  A5      : 0x3ffb21e5  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x0000003c  A9      : 0x3ffb2150  
A10     : 0x3ffc3aa4  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3ffc3d3f  
A14     : 0x3ffc3cfe  A15     : 0x00000005  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000004c  LBEG    : 0x4008427d  LEND    : 0x40084285  LCOUNT  : 0x00000027  

Backtrace: 0x401757c6:0x3ffb2120 0x400d617d:0x3ffb2140 0x400d6191:0x3ffb2170 0x400d61c9:0x3ffb2190 0x400d5fe1:0x3ffb21b0 0x400dcfbd:0x3ffb2210 0x400ddebd:0x3ffb2230 0x400d5e32:0x3ffb2250 0x400e4264:0x3ffb2270 0x4008e812:0x3ffb2290

ELF file SHA256: 29d80fee6

PC: 0x401757c9:  is in Adafruit_GenericDevice::readRegister(unsigned char*, unsigned char, unsigned char*, unsigned short) (c:\Users\stein\Documents\Arduino\libraries\Adafruit_BusIO\Adafruit_GenericDevice.cpp:65).
EXCVADDR: 0x0000004c

Decoding stack results
0x401757c6:  is in Adafruit_GenericDevice::readRegister(unsigned char*, unsigned char, unsigned char*, unsigned short) (c:\Users\stein\Documents\Arduino\libraries\Adafruit_BusIO\Adafruit_GenericDevice.cpp:64).
0x400d617d:  is in Adafruit_BusIO_Register::read(unsigned char*, unsigned char) (c:\Users\stein\Documents\Arduino\libraries\Adafruit_BusIO\Adafruit_BusIO_Register.cpp:250).
0x400d6191: Adafruit_BusIO_Register::read() at c:\Users\stein\Documents\Arduino\libraries\Adafruit_BusIO\Adafruit_BusIO_Register.cpp:189
0x400d61c9:  is in Adafruit_BusIO_RegisterBits::write(unsigned long) (c:\Users\stein\Documents\Arduino\libraries\Adafruit_BusIO\Adafruit_BusIO_Register.cpp:343).
0x400d5fe1:  is in Adafruit_MCP23XXX::digitalWrite(unsigned char, unsigned char) (c:\Users\stein\Documents\Arduino\libraries\Adafruit_MCP23017_Arduino_Library\src\Adafruit_MCP23XXX.cpp:121).
0x400dcfbd:  is in Nexus4Led::setState(LedDefinition, LedState, bool) (c:\Users\stein\Documents\Arduino\libraries\Nexus4/Nexus4Led.h:116).
0x400ddebd: Nexus4Comm::doLoop() at c:\Users\stein\Documents\Arduino\libraries\Nexus4\Nexus4Led.h:102
0x400d5e32: loop() at C:\Users\stein\Documents\Arduino\Nexus-4\Nexus4NewLib\Nexus4NewLib.ino:81
0x400e4264: loopTask(void*) at C:\Users\stein\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.3\cores\esp32\main.cpp:74
0x4008e812: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa\port.c:139
`